zephyrproject-rtos/zephyr #7409: Networking examples crash on optimization levels different than -Os

rlubos, 2018-05-08:
I've encountered a system crash when I was trying to run several networking examples with non-default optimization level on `qemu_x86` target. Every single example that I've tried, after changing optimization level crashed (`coap_client`, `echo_client`, `mbedtls_ssl_client`, `sockets/http_get`). Crash happens on both `Optimize debugging experience` and `Optimize nothing` levels, immeadetly after application starts. I did some initial debugging on `http_get` and from what I've seen, the application did not even enter `main`. Here's the error I've got on qemu:

```
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32,+nx,+pae
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

***** Stack Check Fail! *****
Current thread ID = 0x00402b00
eax: 0x00000080, ebx: 0xaaaaaaaa, ecx: 0x00000000, edx: 0x00010000
esi: 0xaaaaaaaa, edi: 0x0041274c, ebp: 0x0002ad23, esp: 0x00402bd0
eflags: 0x004127a0 cs: 0x2b148
call trace:
eip: 0x00402bd0
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x0000020a
PDE: 0x025 Present, Read-only, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
Current thread ID = 0x00402b00
eax: 0x004127a0, ebx: 0x00000080, ecx: 0x0002b148, edx: 0x00402bd0
esi: 0x0041274c, edi: 0x0002ad23, ebp: 0x004127ac, esp: 0x00412764
eflags: 0x00412760 cs: 0x0008
call trace:
eip: 0x00004c44
     0x000050e8 (0x206)
Fatal fault in essential thread! Spinning...
Terminate emulator due to fatal kernel error
make[3]: *** [zephyr/CMakeFiles/run.dir/build.make:60: zephyr/CMakeFiles/run] Error 1
make[2]: *** [CMakeFiles/Makefile2:341: zephyr/CMakeFiles/run.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:348: zephyr/CMakeFiles/run.dir/rule] Error 2
```

pfalcon, 2018-05-10:
I'd say that the main rule is "the default configuration in the tree should work correctly". Stepping away from that (starting to tweak settings), it's pretty to get to a situation when something doesn't work as expected, or plain break.

I'd say just blindly bumping sizes up isn't good strategy - it may make Zephyr look less competitive that alternatives (and people are going to compare Zephyr with alternatives).

But if we have a good userstory behind a particular optimization setting, it's different matter. For example, -O0 is a distinguished setting, because it's used for code coverage (and can be easily used for debugging too).

jukkar, 2018-05-11:
> As the error indicated, the problem was with stack sizes. Increasing both, main thread stack and idle thread stack let the application run.
> 
> I have a question though, shouldn't default stack sizes be valid for every supported optimization level? What's the approach here, minimal size or versatility?

The stack sizes are always a headache. The default values for various stack sizes work ok normally but if one tweaks things the results could be what you see here. I would not want to increase the stack sizes for some corner case as user could do it by himself anyway by tweaking the kconfig option if needed. Of course it can be difficult to figure out the reason for some weird crash, perhaps some more checks could be added in proper places.

IMHO best option would be if we could get rid of the stack size options (or most of them anyway) from config files. I wonder if it would be possible to figure out how much stack would be needed and the system could suggest a proper value or set it automatically. Anyway, that is quite advanced stuff and would need more investigation.

nashif, 2018-05-11:
based on the above discussion, lowering priority

jukkar, 2019-02-21:
This looks like stale issue, please re-open if it is still valid.

closed 2019-02-21 by jukkar