zephyrproject-rtos/zephyr #7412: Mismatch between 'uint32_t' (and `off_t`) definitions in minimal libc and newlib

oliviermartin, 2018-05-08:
`uint32_t` is defined to `unsigned int` by [minimal libc](https://github.com/zephyrproject-rtos/zephyr/blob/master/lib/libc/minimal/include/stdint.h#L54) while it is defined on my platform (Nordic nRF52840) as `unsigned long int` by newlib.
While in most cases, `sizeof(unsigned int) == sizeof(unsigned long int)`, GCC complains when printf'ing a `uint32_t` value. Either `%ld` or `%d` must be used in one or the other case.

I could ignore the warning with `-Wno-format` but I would prefer to not do so...

Why testing https://github.com/zephyrproject-rtos/zephyr/pull/6391 on my platform, GCC started to complain about `off_t` that is also defined as `int` by minimal libc and `long` by newlib:

```
zephyr/subsys/fs/nvs/nvs.c: In function ‘_nvs_flash_erase’:
zephyr/subsys/fs/nvs/nvs.c:152:9: error: format ‘%x’ expects argument of type ‘unsigned int’, but argument 6 has type ‘off_t {aka long int}’ [-Werror=format=]
  SYS_LOG_DBG("Erasing flash at %x, len %x", offset, len);
         ^~~~~~~~~~~~~~~
zephyr/subsys/fs/nvs/nvs.c: In function ‘nvs_init’:
zephyr/subsys/fs/nvs/nvs.c:477:9: error: format ‘%x’ expects argument of type ‘unsigned int’, but argument 7 has type ‘off_t {aka long int}’ [-Werror=format=]
  SYS_LOG_INF("write-align: %d, write-addr: %x",
```

Zephyr's `u32_t` is always defined as `unsigned int`. So I could use it instead. But in some case (for instance including external library), what should we do?
And what about `off_t`? Should we not use it in Zephyr? Or is it acceptable to cast when printing its value?

andrewboie, 2018-05-08:
You aren't using the right format modifiers.
https://stackoverflow.com/questions/586928/how-should-i-print-types-like-off-t-and-size-t

oliviermartin, 2018-05-09:
Thanks for the pointer :+1:

closed 2018-05-09 by oliviermartin