zephyrproject-rtos/zephyr #7423: samples: net: echo_client: when using the prj_qemu_x86_tls.conf configuration file, the sample runs failed

xianrenqiu, 2018-05-09:
In zephyr/samples/net/echo_client sample, when using the prj_qemu_x86.conf configuration file, the sample runs successfully, but using the prj_qemu_x86_tls.conf configuration file, the sample runs failed.  Error code is EBUSY(-16, TLS handshake not yet done ?).  Detailed process and debugging information are as follows:
```
# branch info
zephyr branch: zephyr-v1.11.0
net-tools branch: master (commit 17e31bef27ff22f6e85893919030a5409963ef31)

# terminal #1
$ cd net-tools
$ sudo ./loop-socat.sh &
$ sudo ./loop-slip-tap.sh
# tap0 info
tap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.0.2.2  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::a870:87ff:fec6:6114  prefixlen 64  scopeid 0x20<link>
        inet6 2001:db8::2  prefixlen 64  scopeid 0x0<global>
        ether aa:70:87:c6:61:14  txqueuelen 1000  (以太网)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

# terminal #2
$ cd net-tools
$ sudo ./echo-server -i tap0
# output info
IPv4: binding to 192.0.2.2
IPv6: binding to 2001:db8::2
Sockets: UPD IPv4 3 IPv6 4, mcast IPv4 5 IPv6 6, TCP IPv4 7 IPv6 8
Socket 3 binding to 192.0.2.2
Socket 4 binding to 2001:db8::2
Socket 5 binding to 239.192.0.2
Socket 6 binding to ff84::2
Socket 7 binding to 192.0.2.2
Socket 8 binding to 2001:db8::2

# terminal #3
$ cd <zephyr>/samples/net/echo_client
$ mkdir build && cd build
$ cmake -DBOARD=qemu_x86 -DCONF_FILE=prj_qemu_x86_tls.conf ..
$ make run
# output info
Zephyr Shell, Zephyr version: 1.11.0
Type 'help' for a list of available commands
shell> [echo-client] [INF] init_app: Run echo client
[echo-client] [ERR] send_tcp_data: Cannot send IPv6 data to peer (-16)
[echo-client] [ERR] send_tcp_data: Cannot send IPv4 data to peer (-16)
```

xianrenqiu, 2018-05-10:
Sorry, I forgot to write some process in front of it, and have been updated on them.  echo_server with TLS support refer to [net-tools](https://github.com/zephyrproject-rtos/net-tools#using-encrypted-ssl-link-with-echo--programs) ，Is there a problem in the process？

xianrenqiu, 2018-05-14:
@pfalcon I have failed many times in the zephyr-v1.11.0 branch, and the latest submission(commit 03f9f6) can run successfully, but If it runs for a long time , there will be an  CPU Page Fault error，The following is the log：
```
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

***** Booting Zephyr OS v1.11.0-1168-g03f9f6649 *****
Zephyr Shell, Zephyr version: 1.11.99
Type 'help' for a list of available commands
shell> [echo-client] [INF] init_app: Run echo client
[echo-client] [DBG] tcp_received: (0x00401408): IPv6: Sent 227 bytes, received 227 bytes
[echo-client] [DBG] compare_tcp_data: (0x00401408): Compared 227 bytes, all ok
[echo-client] [DBG] tcp_received: (0x00400d08): IPv4: Sent 236 bytes, received 236 bytes
[echo-client] [DBG] compare_tcp_data: (0x00400d08): Compared 236 bytes, all ok
... ... ... 
[echo-client] [DBG] tcp_received: (0x00400d08): IPv4: Sent 536 bytes, received 536 bytes
[echo-client] [DBG] compare_tcp_data: (0x00400d08): Compared 536 bytes, all ok
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000068
PDE: 0x025 Present, Read-only, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
Current thread ID = 0x0040d270
eax: 0x00000000, ebx: 0x00414a70, ecx: 0x00413a80, edx: 0x0040d880
esi: 0x00000000, edi: 0x00000001, ebp: 0x00422f84, esp: 0x00422f54
eflags: 0x00000206 cs: 0x0008
call trace:
eip: 0x00020b59
     0x0001ab59 (0x6)
     0x00016af4 (0x414a70)
     0x00016b44 (0x422fd0)
     0x000268ed (0x414a74)
     0x00026120 (0x40d260)
Fatal fault in thread 0x0040d270! Aborting.
```

jukkar, 2018-05-16:
I do not see any handshake failures but sometimes I have seen similar thread crash. I will investigate the crash further.

linkmeyer, 2018-06-05:
@jukkar , what's the status on this?

rveerama1, 2018-06-06:
I am working on this. I didn't find a fix yet.

rlubos, 2018-06-11:
I think I've faced the same (or at least it looks simmilar) bug with CPU Page Fault error.

I did some debugging, and found out that it may happen that `iface` value can be unset when calling `net_stats_update_per_proto_recv`, here:

https://github.com/zephyrproject-rtos/zephyr/blob/8e89206815acdfceed5b059f2cdb8aca8b360273/subsys/net/ip/connection.c#L999

```
[net/conn] [ERR] net_conn_input: {assert: 'pkt->iface' failed} checkpoint 2
[net/conn] [ERR] net_stats_update_per_proto_recv: {assert: 'iface' failed}
***** CPU Page Fault (error code 0x00000000)
Supervisor thread read address 0x00000094
[net/core] [DBG] net_recv_data: (0x00403b58): prio 0 iface 0x00421b80 pkt 0x004082a0 len 153
PDE: 0x025 Present, [net/core] [DBG] net_recv_data: (0x00403b58): prio 0 iface 0x00421b80 pkt 0x00408390 len 74
Read-only, User, Execute Enabled
PTE: 0x00 Non-present, Read-only, Supervisor, Execute Enabled
Current thread ID = 0x00403b58
eax: 0x00000000, ebx: 0x00000000, ecx: 0x0040eef0, edx: 0x0001ef89
esi: 0x000060a5, edi: 0x00421b80, ebp: 0x0040ef40, esp: 0x0040ef04
eflags: 0x00000246 cs: 0x0008
call trace:
eip: 0x00012156
     0x0000f2fc (0x11)
     0x00006514 (0x4082a0)
     0x00018a82 (0x4082a4)
     0x00001c92 (0x403b40)
Fatal fault in thread 0x00403b58! Aborting.
```

So it looks like the problem is that we rely on `pkt` pointer, while it was already deallocated just before in the callback function, and might be cleared/reallocated (I guess that's what happening, didn't investigate what may happen between packet deallocation and call to `net_stats_update_per_proto_recv`).

A possible fix would be to store packet interface on `net_conn_input` entry, and then use it whenever packet may be already deallocated (I think this call may be affected as well): https://github.com/zephyrproject-rtos/zephyr/blob/8e89206815acdfceed5b059f2cdb8aca8b360273/subsys/net/ip/connection.c#L1032  I've tested this solution and it seems to work. Can submit a PR if we agree that's a proper fix.

rlubos, 2018-06-11:
Have modified it in similar way here?:
https://github.com/zephyrproject-rtos/zephyr/blob/8e89206815acdfceed5b059f2cdb8aca8b360273/subsys/net/ip/connection.c#L999
Perhaps I wasn't clear enough, but that's where it actually crashes (at least for me).

rveerama1, 2018-06-11:
Missed that, testing now...

rlubos, 2018-06-11:
Just for reference, I fixed it this way:
https://github.com/rlubos/zephyr/commit/90e970550822167b664094ad302765f03f73a0a5

BTW, is this example in default config crashing on main stack overflow for you as well?

rveerama1, 2018-06-11:
yes, we need to increase main stack size.

jukkar, 2018-06-11:
> Just for reference, I fixed it this way:
> rlubos/zephyr@90e9705

Look very sane thing to do, could you send a PR for this.

rlubos, 2018-06-11:
> Look very sane thing to do, could you send a PR for this.

Sure, I will. Shall I mark that it fixes this issue? I'm not sure if you're observing any other problems with the example?

rveerama1, 2018-06-11:
Running client from past 20 minutes, no crash so far. Mark it as fixed. Include a patch to update stack size also. Thanks.

PR #8312 (2018-06-11): net: samples: fixes for echo_client
Functions for per-interface statistics collection used a pointer to a packet that could've been deallocated in the `net_conn` callback function. In result, application could crash when interface related to the packet was referenced.
To fix that, packet interface is stored earlier, so it can be used instead for statistics collection.

Additionally, increase main stack size for `echo_client` in `prj_qemu_x86.conf`, as the default one caused stack overflow during initialization.

Fixes #7423
closed 2018-06-11 by nashif