zephyrproject-rtos/zephyr #7434: bluetooth: host: sample applications can't set BT address w/o using an FS

mike-scott, 2018-05-09:
Previous to commit d22b7c9f2d771bf9a2a1b8e0378795d4c507f558, sample applications could use the bt_storage API to set the BT address.  That API was removed eliminating functionality that can be useful.

Here are some use-cases:
- Using customized logic to establish BT addresses allows CI testing / management to establish pre-defined values for whitelists / blacklists, etc
- As the BT address is directly translated into a portion of the IPv6 6lowpan address, we can customize the subnet used.  Establishing subnetted networks during test (IE: custom 5th/6th octet values + 32 bit serial number value):
BT Address can be hardcoded to D4:E7 + <32 bit serial # value>.

Example: Calculated 32-bit serial # b1830106 would translate to BT address: D4:E7:B1:83:01:06 which would translate to 6lowpan IPv6 address: fe80::d4e7:b1ff:fe83:106.

The 6lowpan gateway could use the fe80::d4e7/80 mask for networking instead of the typical fe80::/64 mask if desired.

jhedberg, 2018-05-09:
One solution would be to introduce something like the following that needs to be called before bt_enable():

```
int bt_set_id_addr(const bt_addr_le_t *addr);
```

We can discuss whether it should be behind a new Kconfig (defaulting to `n`) or always available.

mike-scott, 2018-05-09:
This would allow callers to change the value of bt_dev.id_addr?

mike-scott, 2018-05-09:
I can't see any reason why it would need to be behind a new Kconfig, but I might be overlooking something.  The old method was available to everyone by default.

carlescufi, 2018-05-09:
If it's runtime then I don't see the point for such a small call to be behind Kconfig, no.

I assume no one likes to extend functions by doing something like `bt_enable_ex(id_addr, cb)` right?

jhedberg, 2018-05-09:
> This would allow callers to change the value of bt_dev.id_addr?

Correct.

> I can't see any reason why it would need to be behind a new Kconfig, but I might be overlooking something. The old method was available to everyone by default.

Fair enough.

> I assume no one likes to extend functions by doing something like bt_enable_ex(id_addr, cb) right?

I was thinking about that, and even about breaking the bt_enable() API by just adding a parameter. However, I'm slightly inclined to not giving this new feature too much publicity since it's for a quite small set of use-cases, and we don't want people misunderstanding or misusing the API (e.g. not understanding what it is and thinking they have to call it no matter what)

carlescufi, 2018-05-09:
> I was thinking about that, and even about breaking the bt_enable() API by just adding a parameter. However, I'm slightly inclined to not giving this new feature too much publicity since it's for a [...]

I agree. I think a `bt_set_id_addr()` is better but let's restrict it to being callable just before `bt_enable()` is invoked in my opinion.

PR #7455 (2018-05-10): Bluetooth: Introduce new bt_set_id_addr() API
There are certain use cases where the application needs to be able to
explicitly set a specific identity address. This was previously
possible using the bt_storage API, however now that it's gone another
solution is needed.

This patch adds a ne bt_set_id_addr() API which the application can
use to set a specific identity address before calling bt_enable().

Fixes #7434
commit 5b01cb1a (2018-05-10): Bluetooth: Introduce new bt_set_id_addr() API

There are certain use cases where the application needs to be able to
explicitly set a specific identity address. This was previously
possible using the bt_storage API, however now that it's gone another
solution is needed.

This patch adds a ne bt_set_id_addr() API which the application can
use to set a specific identity address before calling bt_enable().

Fixes #7434
closed 2018-05-10 by carlescufi