zephyrproject-rtos/zephyr #7437: Zephyr's mailing list archives were hard to find

ulfalizer, 2018-05-09:
I went to `About` -> `Contact us` -> `Mailing lists` on https://www.zephyrproject.org/, which lands you at https://lists.zephyrproject.org/g/main.

That page gives you the impression that the mailing lists are dead, as there are just five messages for April listed. You have to click on `Subgroups` to actually get to the mailing lists. I only kept looking because I already knew there were active mailing lists.

Here's what I'd do:

  1. Move the mailing list link to the top of zephyrproject.org. The IRC link would be nice to have there too.

  2. Have the link go directly to https://lists.zephyrproject.org/g/main/subgroups.

Maybe I'm just bad at looking, but other people probably are too in that case.

dbkinder, 2018-05-09:
I agree.  This happened when the Linux Foundation changed the mailing list hosting software.  I've fixed the link you mentioned to directly to the subgroups page.

dbkinder, 2018-05-09:
Fixed.

closed 2018-05-09 by dbkinder

ulfalizer, 2018-05-09:
Great, thanks!