zephyrproject-rtos/zephyr #7441: newlib support in zephyr is untested and very broken

andrewboie, 2018-05-09:
In the course of working on #6814 I discovered that our problems with this alternate C library run far deeper.

If I modify qemu_x86_defconfig to set CONFIG_NEWLIB_LIBC=y, and then run sanitycheck for qemu_x86, I get many failures:

```
total complete:  282/ 282  100%  failed:  143
139 of 282 tests passed with 0 warnings in 308 seconds
```

nashif, 2018-05-17:
with qemu_x86_nommu all tests pass, could this be related to mmu support and userspace?
qemu_cortex_m3 fails with a few tests that we should be able to fix easily.

nashif, 2018-07-20:
seems like a stack probelm

nashif, 2019-02-23:
we have various dealing with newlib integration, this seems to be a catch-all. Lets deal with those issues individually.

closed 2019-02-23 by nashif