zephyrproject-rtos/zephyr #7442: menuconfig should perform fuzzy string matching for symbols

lpereira, 2018-05-09:
This would make searching for a symbol a whole lot easier, without having to know its precise spelling.  There are many options out there, including shelling out to [fzf](https://github.com/junegunn/fzf) if it's in the path, or [implementing our own](https://medium.com/@Srekel/implementing-a-fuzzy-search-algorithm-for-the-debuginator-cacc349e6c55).

ulfalizer, 2018-05-09:
Yeah, I added regex search already in the first version after I noticed that a lot of searches get awkward without them. With some `.*` and the like sprinkled in, you can usually narrow down what you're looking for.

I wonder how far you would get with a dead simple scheme here:

Since spaces can't appear in symbol names, you could have "foo bar" match entries that match both "foo" and "bar" (which could be regexes too). That way, you wouldn't have to guess which part appears first.

It'd be nice if arbitrary regexes still work. Something too magic might break that.

Another thing I've been thinking of is searching prompts and maybe help texts in addition to symbol names. Not sure if it should always search them all. Could toggle separate modes with some key as well...

ulfalizer, 2018-05-10:
@lpereira 
Added some improvements in https://github.com/zephyrproject-rtos/zephyr/pull/7467.

Still gotta press space, but maybe we can live with that for now. :)

ulfalizer, 2018-05-11:
@lpereira 
Fine with closing this now that https://github.com/zephyrproject-rtos/zephyr/pull/7467?

Might stay open for a long time otherwise, though it's possible that I'll change how the search works at some point.

closed 2018-05-11 by lpereira