zephyrproject-rtos/zephyr #7452: nRF52, NXP kinetis, ARM Beetle and  STM `MPU` option apear for every MPU-equipped device

nvlsianpu, 2018-05-10:
Such options appear in menuconfig under [ARM Options]->[SoC Options]->*

Looks like it is required to make them dependent on appropriate architectures is only reqired for fix this issue.

nvlsianpu, 2018-05-10:
fix for nRF52xx is included in https://github.com/zephyrproject-rtos/zephyr/pull/7449

nvlsianpu, 2018-05-10:
cc @MaureenHelm @galak @nashif @fvincenzo 

I don't have enough knowledge (about device families) for fix this for particular devices - what i can suggest is that it has to by fixed by appropriate maintainers.

galak, 2018-05-10:
Can you explain this a bit more, how are things show up in menuconfig?

nvlsianpu, 2018-05-10:
I added the snipped.

galak, 2018-05-10:
Thanks, I've got a fix I think across all the SoCs.  Will post in a minute.

PR #7457 (2018-05-10): arch: arm: soc: Cleanup Kconfig inclusion per SoC
When we included SoC specific Kconfig files we do something like:

gsource "arch/arm/soc/arm/*/Kconfig.soc"

However, we should do that inside an if block for that SoC family so as
to limit SoC family specific options to only being visible when that SoC
family is selected.  So we simply move the gsource and related items
inside the if SOC_FAMILY_<SOC> block.

Fixes: #7452
commit b493f484 (2018-05-10): arch: arm: soc: Cleanup Kconfig inclusion per SoC

When we included SoC specific Kconfig files we do something like:

gsource "arch/arm/soc/arm/*/Kconfig.soc"

However, we should do that inside an if block for that SoC family so as
to limit SoC family specific options to only being visible when that SoC
family is selected.  So we simply move the gsource and related items
inside the if SOC_FAMILY_<SOC> block.

Fixes: #7452
closed 2018-05-10 by galak