zephyrproject-rtos/zephyr #7460: Unable to view PR #6391

dbkinder, 2018-05-10:
https://github.com/zephyrproject-rtos/zephyr/pull/6391 fails to open and shows the "unicorn" page (page taking too long to load).  I reported this to the GitHub folks.  Are others having the same problem?

dbkinder, 2018-05-10:
GitHub replied: "We've got an issue open internally for our engineers to check out that PR to see what might be going on. We'll update you as soon as we know more.  Cheers, Alex"

nashif, 2018-05-11:
it just has too many comments and content, this has been very common with PRs that have >100 comments

dbkinder, 2018-05-11:
If we can't view it we can't review it and it's stuck open?

nashif, 2018-05-11:
yes, this needs to be fixed on github side, but I can see it an review it by not using the browser.

henrikbrixandersen, 2018-05-13:
FWIW, I can view #6391 fine using Safari on macOS.

pswarnak, 2018-05-14:
I can view it using chrome browser. Possibly, the issue is fixed now. Please check.

MaureenHelm, 2018-05-14:
I got the unicorn too. Chrome and IE11 on Win10.

MaureenHelm, 2018-05-14:
Tagged the priority as low so it doesn't block the zephyr 1.12 release. In reality it's a higher priority than that, but it's not something we can fix ourselves.

nashif, 2018-05-15:
PR is now merged.

closed 2018-05-15 by nashif