zephyrproject-rtos/zephyr #7478: tests: valgrind detected issues

aescolar-ot, 2018-05-11:
Several issues reported by valgrind on tests

## Control depends on uninitialized variables:

### tests/kernel/mbox/mbox_api/kernel.mailbox
```
starting test - test_mbox_block_get_invalid_pool
==21984== Thread 2:
==21984== Conditional jump or move depends on uninitialised value(s)
==21984==    at 0x8056B1F: mbox_message_match (mailbox.c:143)
==21984==    by 0x80579EF: k_mbox_get (mailbox.c:440)
==21984==    by 0x804AE23: tmbox_get (test_mbox_api.c:395)
==21984==    by 0x804B95F: tmbox_entry (test_mbox_api.c:538)
==21984==  Uninitialised value was created by a stack allocation
==21984==    at 0x804A4DE: tmbox_get (test_mbox_api.c:311)
==21984== 
==21984== Conditional jump or move depends on uninitialised value(s)
==21984==    at 0x80577DC: mbox_message_data_check (mailbox.c:411)
==21984==    by 0x8057A78: k_mbox_get (mailbox.c:447)
==21984==    by 0x804AE23: tmbox_get (test_mbox_api.c:395)
==21984==    by 0x804B95F: tmbox_entry (test_mbox_api.c:538)
==21984==  Uninitialised value was created by a stack allocation
==21984==    at 0x804A4DE: tmbox_get (test_mbox_api.c:311)
==21984== 
PASS - test_mbox_block_get_invalid_pool
```
### tests/kernel/critical/kernel.common
```
==23751== Conditional jump or move depends on uninitialised value(s)
==23751==    at 0x80491F9: atomic_and (atomic.h:250)
==23751==    by 0x80492F9: atomic_clear_bit (atomic.h:391)
==23751==    by 0x804936A: k_work_init (kernel.h:2253)
==23751==    by 0x8049799: critical_loop (main.c:87)
==23751==    by 0x8049A06: regression_task (main.c:152)
==23751==  Uninitialised value was created by a stack allocation
==23751==    at 0x8049721: critical_loop (main.c:80)
==23751== 
==23751== Conditional jump or move depends on uninitialised value(s)
==23751==    at 0x80491B7: atomic_or (atomic.h:208)
==23751==    by 0x804926F: atomic_test_and_set_bit (atomic.h:371)
==23751==    by 0x80493EF: k_work_submit_to_queue (kernel.h:2282)
==23751==    by 0x80497C9: critical_loop (main.c:88)
==23751==    by 0x8049A06: regression_task (main.c:152)
==23751==  Uninitialised value was created by a stack allocation
==23751==    at 0x8049721: critical_loop (main.c:80)
==23751== 
==23751== Thread 6:
==23751== Conditional jump or move depends on uninitialised value(s)
==23751==    at 0x8059856: atomic_and (atomic.h:250)
==23751==    by 0x80598D1: atomic_test_and_clear_bit (atomic.h:350)
==23751==    by 0x8059A73: work_q_main (work_q.c:37)
==23751==  Uninitialised value was created by a stack allocation
==23751==    at 0x8049721: critical_loop (main.c:80)
```
### drivers/crypto/test-micro
```
==24446== Use of uninitialised value of size 4
==24446==    at 0x804E55B: inv_sub_bytes (aes_decrypt.c:109)
==24446==    by 0x804E7EE: tc_aes_decrypt (aes_decrypt.c:148)
==24446==    by 0x804FBE5: tc_cbc_mode_decrypt (cbc_mode.c:106)
==24446==    by 0x8057107: do_cbc_decrypt (crypto_tc_shim.c:58)
==24446==    by 0x8049AEA: cipher_cbc_op (cipher.h:219)
==24446==    by 0x804A617: cbc_mode (main.c:179)
==24446==    by 0x804B675: zephyr_app_main (main.c:420)
==24446==    by 0x805872A: bg_thread_main (init.c:254)
==24446==  Uninitialised value was created by a stack allocation
==24446==    at 0x804A15F: cbc_mode (main.c:115)
==24446== 
[general] [INF] cbc_mode: CBC mode DECRYPT - Match
```

lpereira, 2018-05-17:
While these warnings might be indeed issues, Valgrind warnings have to be taken with a grain of salt.

For instance, the last one, where it says that an uninitialized value was created by a stack allocation: a `struct cipher_ctx` is indeed created in the stack, uninitialized, and then a few items that are used by some of the TinyCript functions are called.  Apparently only the things that were initialized are used, so it's fine;  the struct, however, still contains uninitialized values.  The warning will go away if it's initialized with designated initializers: `struct cipher_ctx ini = {.keylen=16, .key.bit_stream = key, .flags = cap_flags}`.

I didn't check the others.

nashif, 2018-05-26:
@lpereira can you please look at the others?

lpereira, 2018-05-29:
I'm hitting this [issue here](https://bugzilla.redhat.com/show_bug.cgi?id=1566639); using the updated valgrind package doesn't seem to get me anywhere.

Regardless, I'm taking a closer look at these issues.  The one I looked already is presumably fixed.

lpereira, 2018-05-29:
Sent a PR fixing these issues.

PR #8004 (2018-06-05): Address some issues found with Valgrind
Please refer to #7478 for details.

I'm not really sure about the one fixing the mbox test; the warning is gone, but I don't know if a `struct k_mbox_msg` should be fully zeroed out prior to a `k_mbox_get()` call.

The k_work fix doesn't have a call to `atomic_clear_bit()` any longer; this shouldn't be an issue when setting its value to `0`, but I'd like a second opinion on this.

Fixes #7478
closed 2018-06-05 by nashif