zephyrproject-rtos/zephyr #7480: pthread_attr_init( ) does not conform to POSIX specification

aescolar-ot, 2018-05-11:
Currently `pthread_attr_init()` checks the content of the input pthread_attr.
Therefore if its input was not initialized before the call, the behavior of the function is unpredictable.
Note that the point of pthread_attr_init() is to actually initialize it after declaring it and before using it.

Also, currently the function may return EBUSY which is not a valid response for `pthread_attr_init()`.

The POSIX spec. for this function can be found here:
http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_attr_init.html

An appropriate behavior of the function would be to always copy into its input the init structure without checks. As the specification allows undefined behavior for `pthread_attr_init()` if the input had been initialized before hand.

Related: #7446

aescolar-ot, 2018-05-11:
@nashif would you be so kind as to reassign it to whoever maintains this part of the code?

nashif, 2018-05-14:
@ramakrishnapallala please take a look at this.

PR #7562 (2018-05-16): lib: posix: Fix pthread_attr_init() return code
pthread_attr_init() should not return EBUSY as per POSIX spec
so fixed this by return ENOMEM if the attr pointer is NULL.

Also fixed the attribute initialization logic by copying the
init_pthread_attrs to the attr.

Fixes #7480
closed 2018-05-16 by nashif