zephyrproject-rtos/zephyr #7499: [RFC] Get rid of Zephyr's prefer-later-defaults behavior

ulfalizer, 2018-05-12:
Zephyr adds a [patch](https://github.com/zephyrproject-rtos/Kconfiglib/commit/69c5735dad2d5ded9768634f1aa0c8cb283accb7) to vanilla Kconfig that makes it prefer later defaults, as documented [here](http://docs.zephyrproject.org/porting/board_porting.html#zephyr-specific-kconfig-behavior-for-defaults).

I think that change was made so that [`Kconfig.defconfig` files](http://docs.zephyrproject.org/porting/board_porting.html#configuring-invisible-kconfig-symbols) can override defaults in the "base" configuration. They are included last.

I think the standard Kconfig behavior would be nicer, which is to use the first default with a satisfied condition. It's what people expect coming from other projects that use Kconfig, and you wouldn't have to carry that patch around.

I thought of a "safe" way to make the transformation:

1. Hack Kconfiglib to print defaults in reverse order when printing symbols. Print all symbols and save the output somewhere.

2. Include the `Kconfig.defconfig` files first instead of last, and switch the order of all the defaults around. Maybe other includes would need to be switched around as well.

3. Remove the Kconfiglib hack, print all symbols, and check that the output is still the same.

It would still take some boring manual work, but it seems doable.

Any objections? After that change, Zephyr would use vanilla Kconfig.

At some point, it might be possible to install Kconfiglib and the menuconfig via `pip` instead of having them in the source tree.

ulfalizer, 2018-05-12:
CC @carlescufi @SebastianBoe @mbolivar

nashif, 2018-05-13:
as long as you have a way to verify that the change does not impact the generated configurations and does generate the same final .config we should be fine with such change. Need to test this basically on all boards and tests and check if there are any differences.

ulfalizer, 2018-05-13:
@nashif 
Should be fine as long as the defaults are evaluated in the same order as before. I'll check with some of the examples too.

Thought I'd do what `doc/Makefile` does to include everything when comparing outputs.

carlescufi, 2018-05-13:
>Any objections?

None from me, but as @nashif said, we need to make completely sure we haven't modified the generated `.config` files by running some sort of `diffconfig` on all `.config` files generated by `sanitycheck`.

>gsource should usually be avoided as well

Why? I am not sure why globbing is a bad idea. We also use it in the current work to split Zephyr into multiple repositories, as you can see here:

https://github.com/zephyrproject-rtos/zephyr/pull/7338/files#diff-d6f4145321ef0d57a017a2079114715dR41

ulfalizer, 2018-05-13:
@carlescufi 
Just some personal anti-globbing bias. Can't see at a glance what (isn't) being included, plus you might accidentally put something in there that matches (and yeah, that's actually happened to me in other projects).

The Python blowhard version would be "explicit is better than implicit" I think. :D

Maybe it's not a static set though, and yeah, just some personal experience. No biggie.

ulfalizer, 2018-05-14:
Working on https://github.com/zephyrproject-rtos/zephyr/issues/6879 now. Getting that fixed will help with some copy-paste action here.

mbolivar, 2018-05-16:
> Any objections?

What about out of tree applications? We currently specifically recommend applications put their content after including the Zephyr config file:

http://docs.zephyrproject.org/application/application.html#application-cmakelists-txt

This reversal would probably mean they should put their config options (and any default settings they want to override Zephyr's with) before including Zephyr's root Kconfig, no?

If so, this is a breaking change that should be communicated clearly to users.

ulfalizer, 2018-05-17:
@mbolivar 
They'd also need to reverse the defaults on all symbols that have more than one default, so could get messy if there's a lot of out-of-tree stuff.

mbolivar, 2018-05-17:
@ulfalizer hmm, "messy" makes me a little nervous :).

It makes sense to get back to the standard Kconfig behavior, though.

Is it possible to default to the standard behavior, but emit a message that it has changed, and letting apps define a CMake variable before including the boilerplate file that restores the "odd" behavior?

We could keep that for a deprecation period of say one release, before removing the odd behavior entirely.

I also want to confirm that by defining a default before including Zephyr's Kconfig root, they could override any default provided by the build (this is important when setting defaults conditionally).

mbolivar, 2018-05-22:
> Maybe some temporary ugly hack that switches stuff around in some way if you put a flag in the Kconfig file would be possible, if needed...

That's what I'm saying, basically.

If the user runs "cmake -DKCONFIG_USE_WEIRD_ZEPHYR_BEHAVIOR", then kconfiglib gets passed some option that enables the existing behavior.

Otherwise, it reverts to the "normal" Linux kernel behavior, but emits a warning that there was a breaking change (perhaps through a second option, so as not to pollute output when  run in its primary use case).

This at least gives app developers a sporting chance to notice what happened and fix any issues, especially if the warning is high quality and offers hints on what to do in case things go wrong.

carlescufi, 2018-12-09:
@ulfalizer I think we can close this now, agree?

ulfalizer, 2018-12-09:
Yups, fine to close.

closed 2018-12-09 by ulfalizer