zephyrproject-rtos/zephyr #7500: DHCP: when CONFIG_NET_APP_MY_IPV4_ADDR is IP stack needs to rebind on DHCP aquire

mike-scott, 2018-05-12:
When testing DHCP w/ LwM2M client sample which sets CONFIG_NET_APP_MY_IPV4_ADDR="192.0.2.1" to work with QEMU by default, it was noticed that bind was happening on 192.0.2.1 and incoming packets on the DHCP address were being ignored.

We need to rebind anytime IP changes to avoid this.

mike-scott, 2018-05-12:
Reported by @therealprof as a part of debugging https://github.com/zephyrproject-rtos/zephyr/issues/7475

jukkar, 2018-05-14:
I could not reproduce this, or then I misunderstood the description.
I did following to test this (several terminals are needed):

Setup the qemu_x86 terminal for SLIP/TAP:

```
cd net-tools
./loop-socat.sh
sudo ./loop-slip-tap.sh
```

Terminal for DHCP server:
```
cd net-tools
sudo ./dnsmasq.sh
```

Terminal for echo-server:
```
cd samples/net/echo_server
cp prj.conf prj.conf.dhcpv4
echo "CONFIG_NET_DHCPV4=y" >> prj.conf.dhcpv4
mkdir build && cd build
cmake .. -DBOARD=qemu_x86 -DCONF_FILE=prj.conf.dhcpv4
make run
```

The zephyr instance in qemu will first set the static IPv4 address and then after a while the address will be overridden by the DHCP address. Of course if there is a request before DHCP address is received, then wrong address (the static one) is used for a short period. One option to solve this would be to wait NET_EVENT_IPV4_ADDR_ADD event and continue only after NET_ADDR_DHCP address type is set.

therealprof, 2018-05-14:
> Of course if there is a request before DHCP address is received, then wrong address (the static one) is used for a short period.

The "short period" part is not correct. If an application binds to a port before the DHCP address arrives then all packets to the new address will be dropped (and an IGMP unreachable message generated), eternally.

> One option to solve this would be to wait NET_EVENT_IPV4_ADDR_ADD event and continue only after NET_ADDR_DHCP address type is set.

That doesn't solve the problem. The IP addresses may change at any time and the system needs to be able to cope with that situation.

Typically systems support binding to IP **and** port or just port. The former is what seems to happen at the moment and the latter is what should usually should happen.

jukkar, 2018-05-14:
> Typically systems support binding to IP and port or just port. The former is what seems to happen at the moment and the latter is what usually should happen.

That sounds like a application issue (or could be in net_app) where IP address is set instead of just the port when creating the connection listener.

jukkar, 2018-05-14:
> The "short period" part is not correct. If an application binds to a port before the DHCP address arrives then all packets to the new address will be dropped (and an IGMP unreachable message generated), eternally.

That is why applications should listen IP address add/del events and act accordingly.

mike-scott, 2018-05-14:
@jukkar I have a feeling this is net_app specific.  But, I'll look through the LwM2M client code to be sure.

mike-scott, 2018-05-14:
Adding @pfalcon to this issue.

I've traced this issue into the IP stack where we *change* the IP address of an existing interface where the unicast addr_type is set to: NET_ADDR_OVERRIDABLE and the incoming address type is NET_ADDR_DHCP:
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/net_if.c#L1965

In this case, we may have called net_conn_register() on the existing IP address and it will need to be removed / updated.

mike-scott, 2018-05-14:
A debug of FRDM-K64F connecting to public Leshan LwM2M server: https://hastebin.com/gogicuruwi.go

Highlights from the log:

[net/app] [INF] setup_ipv4: IPv4 address: 192.0.2.1 (using the static IP set in CONFIG_NET_APP_MY_IPV4_ADDR)
[net/ctx] [DBG] net_context_bind: (0x20003a58): Context 0x20001c94 binding to UDP 192.0.2.1:60843 iface 0x20008380 (called from _net_app_set_net_ctx() in net_app.c)
[net/conn] [DBG] net_conn_register: (0x20003a58): [1/2/17/0x33] remote 0x20001ca4/5.39.83.206/5683 local 0x20006930/192.0.2.1/60843 cb 0x000063d1 ud 0x20000000 (connection.c registering the connection)
...
[net/app] [INF] ipv4_addr_add_handler: IPv4 address: 192.168.69.145 (DHCP request is answered and new IP is assigned)
...
[lib/lwm2m_rd_client] [DBG] sm_send_registration: registration sent [5.39.83.206]
[net/conn] [DBG] net_conn_input: (0x20001ef0): Check UDP listener for pkt 0x200046d0 src port 5683 dst port 60843 family 2 chksum 0xa78b len 46
[net/conn] [DBG] net_conn_input: (0x20001ef0): No match found.

Responding Leshan server is rejected because the old IP 192.0.2.1 is still in the conns[] array, not the new IP.

This is most definitely an event that should be handled in the IP stack and not by each sample.

therealprof, 2018-05-15:
> Sounds like we would be able to fix this by just making sure that we bind to only port and not address, then it would not matter if we change the interface address.

Yes, that's what I tried to say. Binding to IP addresses is a special case rather than the norm and I'm not sure we'll ever need that in a MCU based setup.

pfalcon, 2018-05-15:
> or then I misunderstood the description.

I've read thru the available comments, and still not sure I understand the issue. So, let me start with a few facts instead:

1. Support for being able to use both static address (CONFIG_NET_APP_MY_IPV4_ADDR) and DHCPv4 was added in https://github.com/zephyrproject-rtos/zephyr/pull/5750 , for debugging/demonstration purposes: to let a sample app with the same config to be able to plug both into a host, which requires static IP address, as a host usually doesn't run DHCP server (because it itself a DHCP client), or a router (requires DHCP). Such a mode is not intended to be used in production, as an extension to the recommendation to not use static IP addresses in production (due to conflicts, etc.). So, if your, @mike-scott and @therealprof, concerns are related to production usage, I'd suggest to not rely on CONFIG_NET_APP_MY_IPV4_ADDR in the first place.

2. I did test that frdm_k64f works per the requirements above: when plugged into my laptop, I was able to access a server running on it via static address. Likewise, when plugged into my router, I was able to look up DHCP address assigned to it and access the server by it.

I'll try to retest p.2 today. (For reference, IIRC I tested it with a simple webserver written in MicroPython. Today I'm going to test with samples/net/sockets/dumb_http_server, which is my default testing app).

jukkar, 2018-05-15:
> Sounds like we would be able to fix this by just making sure that we bind to only port and not address, then it would not matter if we change the interface address.
>
> Yes, that's what I tried to say. [...]

For servers we typically bind using just a port. But this issue was about client in which case we need to figure out the local address that is used when sending the data.

Anyway, I could reproduce this issue with echo-client in qemu. I have a "fix" for selecting the source address properly although I am not sure if it fixes exactly this issue. I will send the PR shortly for testing.

pfalcon, 2018-05-15:
Few more comments:

> and I'm not sure we'll ever need that in a MCU based setup.

I keep hearing this argument, as if "MCU based setup" is so different and new that it requires all the decades of industry's (IP) networking handling redefined. I'd however lean towards a different approach - there's a well defined APIs, semantics in POSIX/BSD Sockets approaches. It should be implemented faithfully. Then it's up to "MCU based apps" authors to implement requirements of their apps using those APIs and semantics.

As an example, what would happen if you, on your Linux system, start to load a webpage , and at this moment DHCP address changes? I'm sure result will be error page from a browser. It's not a problem though - a user will just press "Reload" button instinctively.

How "MCU based apps" differ from this is that there's no user to press "Reload" button. But why don't you think guys that your apps shouldn't do that *themselves* - on the highest level (i.e. level of the app) to reinitialize all networking connections, servers, etc. - in case of networking environment changes?

therealprof, 2018-05-15:
@pfalcon I've no intention to use such a setup in production, I just happened to stumble into it because it is the default configuration of the example application. However I'm also worried (have not tested it, yet) that this might also apply to IP address changes which happen on the fly, e.g. when the DHCP lease expires and a new one is obtained. I also have not tested IPv6 and/or TCP, only UDP.

therealprof, 2018-05-15:
@pfalcon 

> I keep hearing this argument, as if "MCU based setup" is so different and new that it requires all the decades of industry's (IP) networking handling redefined.

That's neither what I said, nor implied. To me it is pretty clear that pretty much **any** stack only implements a **subset** of the available standards and the requirements and choice of implemented subsets varies by application.

> How "MCU based apps" differ from this is that there's no user to press "Reload" button. But why don't you think guys that your apps shouldn't do that themselves - on the highest level (i.e. level of [...]

I think you lost me here. First you're saying the network stack should have all the usual POSIX capabilities but then you're trying to say that this doesn't apply to applications because they can simply reinitialise whenever the ip address or route changes, which is definitely not how applications using regular sockets work.

pfalcon, 2018-05-15:
@therealprof :

> However I'm also worried (have not tested it, yet) that this might also apply to IP address changes which happen on the fly

+1 for that worry. I won't be surprised if the current Zephyr IP stack doesn't handle them. But neither I'm sure that it should handle them, or how other IP stacks deal with various cases like that (stacks of different scale, from embedded, direct competitors of Zephyr, to big systems like Linux, with whose functionality Zephyr won't be able to compete).

That's why I wish there was a general discussion, leading to a set of requirements, how such conditions should be handled - in the IP stack, or left to the apps. And note that at least *some* networking exceptional conditions will need to be handled on apps' level. Then maybe it's not too bad an idea to leave them *all* to apps. If not, we (the IP stack developers) have a lot of work ahead of us ;-).

pfalcon, 2018-05-15:
> Binding to IP addresses is a special case rather than the norm and I'm not sure we'll ever need that in a MCU based setup.

Binding to an IP address is not a special case in any way and is absolutely a norm.

From "main 7 ip":

~~~
       When a process wants to receive new incoming packets or connections, it
       should bind a socket to a local interface address  using  bind(2).   In
       this case, only one IP socket may be bound to any given local (address,
       port) pair.  When INADDR_ANY is specified in the bind call, the  socket
       will  be bound to all local interfaces.  When listen(2) is called on an
       unbound socket, the socket is automatically bound to a random free port
       with the local address set to INADDR_ANY.  When connect(2) is called on
       an unbound socket, the socket is automatically bound to a  random  free
       port  or  to  a  usable  shared  port  with  the  local  address set to
       INADDR_ANY.
~~~

As you may imagine, it's not documented what happens: 1) if specific interface, a socket bound to, changes IP address; 2) for INADDR_ANY case, if one of the interfaces changes address; 3) ...; 4) ... .

We can apply "common sense", e.g. for case 2, it seems that new address should have an effect, while for case 1, not necessarily. But it would be interesting to study how other stacks deal with this, and assess complexity of particular implementations.

P.S. Linux also has a (non-POSIX apparently) way to bind to a particular interface (not knowing it address).

therealprof, 2018-05-15:
@pfalcon

> or how other IP stacks deal with various cases like that

Easy: If you have a socket bound to just a port (rather than a specific IP/port, interface/port or even interface/IP/port) then any change of address is completely transparent to the application, in fact there is not even a standard notification mechanism for address changes.

pfalcon, 2018-05-15:
> Easy: If you have a socket bound to just a port

Per the description above (which is based on POSIX), it's impossible to bind to "just a port".

therealprof, 2018-05-15:
@pfalcon 

> Binding to an IP address is not a special case in any way and is absolutely a norm.

It's a standard functionality but one that does not have any decent use case on an embedded system. IP address binding is done e.g. on servers to get around the number of open sockets problem and it's done on desktops to bind applications to localhost. Those are the two most common uses

> Per the description above (which is based on POSIX), it's impossible to bind to "just a port".

You're reading that wrong: INADDR_ANY means exactly that.

therealprof, 2018-05-15:
> "Any change of address is completely transparent to the application" is a security issue. You can assume that you configured connections in such a way that "management" requests are accepted only on "secure" interface, then something changes, and your management interface is wide open to the world and attackers.

That is very far fetched, but okay. Binding a port to an IP address for security reasons is security by obscurity, same as thinking you don't need a firewall because your address is NATed.

> Doing too much magic behind the back leads to issues...

Absolutely, which is exactly why I don't want any magic.

closed 2018-05-16 by jukkar

pfalcon, 2018-05-17:
> I'll try to retest p.2 today.

Sorry, got backlogged. The test process is documented in https://github.com/pfalcon/zephyr/issues/6 . As a summary, dumb_http_server BSD sockets sample (which is a server app which binds to INADDR_ANY) is able to accepts connections both when DHCP assignment didn't happen and when it did (i.e. it works "as expected").

pfalcon, 2018-05-18:
@mike-scott :

> Currently, the IP stack doesn't support "rebinding" from what I see. That would need to be fixed before taking the next step.

For reference, I came up with an easy way to test how a Linux (or similar OS) deals with the change of interface address when there's a connection open: https://github.com/zephyrproject-rtos/zephyr/issues/7630#issuecomment-389975170 . Based on that experiment, it looks like a network connection's local IP is not dynamically tied to the IP address of the interface the connection was created on.

"Rebinding" issue originally comes from a different matter: whether it's possible to call bind() twice. In Linux, it's possible (IIRC), while Zephyr doesn't support that, which may pose some problems given that a call to bind() may be implicit. (Yes, inability to rebind would affect the matter we discuss, but so far nothing points that we should "rebind" in the IP stack on the intereface address change.)

PR #7544 (2018-05-16): net: app: Select proper source IPv4 address in client
Instead of selecting the first IPv4 address from the network
interface, use destination address to select the proper local IPv4
address.

Fixes #7500
commit e50cacb3 (2018-05-16): net: app: Select proper source IPv4 address in client

Instead of selecting the first IPv4 address from the network
interface, use destination address to select the proper local IPv4
address.

Fixes #7500