zephyrproject-rtos/zephyr #7503: LwM2M client: No registration retry after error?

therealprof, 2018-05-14:
It seems that the LwM2M client aborts any interaction with the server if it receives an unexpected reply, e.g.:
```
[lib/lwm2m_rd_client] [INF] lwm2m_rd_client_start: LWM2M Client: nucleo_f429zi
[lib/lwm2m_rd_client] [INF] sm_do_init: RD Client started with endpoint 'nucleo_f429zi' and client lifetime 30
[lib/lwm2m_rd_client] [DBG] sm_send_registration: registration sent [212.18.24.70]
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [212.18.24.70]
[lib/lwm2m_rd_client] [DBG] do_registration_reply_cb: Registration callback (code:2.1)
[lib/lwm2m_rd_client] [ERR] do_registration_reply_cb: Unexpected endpoint data returned.
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: reply 0x200000d8 handled and removed
```

After that point: silence. Instead I would expect a retry after some time since the error might be temporary and/or sporadic.

NB: I moved the client lifetime check before the output, to make it show the default lifetime instead of 0 if not explicitly set.

mike-scott, 2018-05-14:
@therealprof This is the case where the server is presenting the endpoint data as "rd/<registered end point name>" in 1 LOCATION_PATH option vs. splitting up them up into 2 options ("rd" "<registered end point name").

How long would be a good retry timeout period for errors like this vs. non-response/timeout errors which restart the registration process the next processing cycle (about 1 second)?  I might need to look back at the LwM2M spec to see if they have any guidance here.

There are other such cases where the client basically gives up:
- Receives first portion of the LOCATION_PATH which isn't "rd"
- Receives a response code "not found" indicating the register device endpoint on the server is invalid
- Receives a "forbidden" code indicating the server requires security information that the client isn't presenting.

Should all of these have the same timeout / retry logic?  The idea being we don't want to retry too often in bad situations like these as to place the server under DDOS.

mike-scott, 2018-05-14:
> Usually random pick from exponentially increasing interval.

Sounds good.  I'll setup a patch once my queue settles down a bit.

> If that happens while registering, I would expect the client to go into bootstrap mode.

Hmm, I should look at making this change when my bootstrap support patches land.  I'll keep this issue open till then.  My current local tree has bootstrap support as optional so we would still need to handle a case where bootstrap wasn't supported and we end up in a "bad" state.  Perhaps the exponential backoff works here as well for that purpose.

ioannisg, 2019-07-30:
@therealprof please, re-open if needed, to continue discussion

closed 2019-07-30 by ioannisg