zephyrproject-rtos/zephyr #7511: [Coverity CID :185391] Incorrect expression in /tests/lib/rbtree/src/main.c

mandarcthorat1, 2018-05-14:
Static code scan issues seen in File: /tests/lib/rbtree/src/main.c
 Category: Incorrect expression
 Function: check_tree
 Component: Tests
 CID: 185391
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

spoorthik, 2018-05-25:
The issue is already been fixed. 

The code shown in scan9.coverity:
`CHECK(node_index(walked_nodes[ni++]) == i);`

But the latest version of code is:
```
CHECK(node_index(walked_nodes[ni]) == i);
ni++;
```
This doesn't require any modification. Closing the bug.

closed 2018-05-25 by spoorthik

mandarcthorat1, 2018-05-25:
commit 7e0d1d27d5c6bb3899002fc882413cbf8849e7bd has fixed this issue