zephyrproject-rtos/zephyr #7516: Support binary blobs and glue code

carlescufi, 2018-05-14:
I was wondering how Zephyr feels about upstreaming "glue code" to a binary blob (binary linkable library) upstream.

What I was wondering is if it people think it would be reasonable to add the necessary "glue" changes upstream, which mainly means:

* New Kconfig options to enable a new module or replace an existing one with the binary blob
* Required CMake changes in order to incorporate the library as an option
* The necessary glue code to enable building against the library

I think there are 3 ways we can look at this:

1. Disallow it completely, require vendors to maintain those patches downstream
2. Allow it and not test it, or require the vendor to test it and post results to TestRail
3. Allow it and then require the binary blob to be present in a specific Zephyr-repo in the multi-repo world. This would allow us to actually test it.

galak, 2018-05-14:
I think this is something we need to support in Zephyr.  We probably need to figure out the rules and document them around what is considered an acceptable binary.  For example:

- is the binary licensed in such a way that allows re-distribution by the Zephyr project. 
- are the interface/headers to the binary licensed permissively/OSI licensed (ie, something like Apache or BSD).

carlescufi, 2018-05-14:
>We probably need to figure out the rules and document them around what is considered an acceptable binary

I agree. Where do you think this document should live?

Also, @galak I guess you are assuming option 3. and the multi-repo being there?

galak, 2018-05-14:
> I agree. Where do you think this document should live?

I assume the doc would be part of our contributing guidelines, so something under doc/contribute/

> Also, @galak I guess you are assuming option 3. and the multi-repo being there?

Yes, I think something like option 3.

carlescufi, 2018-05-14:
Depends on #6770

carlescufi, 2018-05-14:
>This sounds somewhat similar to what is done here:

Yes, indeed. I guess that flew under the radar.

dleach02, 2018-05-14:
+1, I'm starting down this path right now to figure out how to integrate support for BLE on the NXP KW41 platforms. The BLE controller code is a binary blob that comes from NXP and won't be able to be hosted on the Zephyr repository due to legal restrictions.

erwango, 2018-05-28:
As an output of this ticket it would be nice to have a checklist that can be can be used to check easily if the component we want to introduced is compatible with zephyr policy and if not what is missing.

nashif, 2018-06-11:
> is the binary licensed in such a way that allows re-distribution by the Zephyr project.

this only applies if we actually are going to support that library as part of the project, in most cases I would say, no, we should not have any binaries included in the project itself, however we should enable this like with the ESP32 case. 
Once we have  multi-git repo support his would be easier obviously but regardless, a policy is still required.

microbuilder, 2019-06-19:
While I would personally argue that no binary objects should be included in the core repo, it's a lot more ambiguous how to handle additional repositories hosted under `zephyrproject-rtos`. 

Ignoring the debate of open source or not, from a practical point of view it seems like this would place a significant legal burden on the Zephyr Project to validate that we **can** host and distribute said binary blob(s) and associated IP or files. 

The opinion of engineer X on Slack or over email saying "sure, it's fine to host blob Y" probably isn't sufficient for blobs that have EULA's associated with them, and would likely require written sign-off by appropriate IP owners. If they are distributing binary blobs, there are almost certainly distribution terms attached.

Some cases may be clear cut, but it seems like there would still need to be some sort of legal validation process or paper trail here to host binaries in the Github org. Whether the benefit is worth the additional effort and legal exposure is obviously open for debate.

pabigot, 2019-06-23:
Please retain the distinction between "Apache-2.0" and "OSS" in this discussion.  They are very much not the same.

I'm not clear on what "OSS" is intended to identify.  It would perhaps be better to use the term [OSI-approved license](https://opensource.org/licenses) where open source beyond Apache-2.0 is being addressed.  Note that GPL is OSI-approved.

pabigot, 2019-06-23:
Repeating here the position I gave in TSC chat on 2019-06-12:

Zephyr bills itself as Apache-2.0, which I believe to be true of the main repository.  Through west it pulls in material from other repositories; at the moment these are hosted by zephyrproject-rtos, but in principle they could be external.  It's not clear to me whether all material in those repositories is also Apache-2.0.

As an open source developer I may need to protect against claims of potential IP theft by doing development on a clean system, where in theory I know the licensing of every bit on the machine.

Question 0 does not appear in the description of this issue:

0. Should a default-installed Zephyr installation be restricted to content that is licensed under Apache-2.0?

I personally prefer the default answer to this to be "Yes."  Whether it is "yes" or "no", the answer (and its rationale) will inform the process leading to the remaining decisions, including whether/how non-Apache-2.0 content can be exposed as an opt-in post-installation step from the default installation, and how the licensing of any retrieved material is recorded and accessed.

When I do `west update` to fetch everything, I have no idea what's just been installed on my system.  This is somewhat concerning to me.

marc-hb, 2019-06-24:
> When I do west update to fetch everything, I have no idea what's just been installed on my system. This is somewhat concerning to me.

+1 and I believe you've been considering only the default `zephyr/west.yml`. West should have a `west list --format={license}` field that works for any manifest. From a developer (and lawyer) perspective it's much more important to know the licenses that are going into a product than which server or mirror it's hosted on.

Examples:
`pip3 install --user pip-licenses && pip-licenses`
https://www.yoctoproject.org/docs/2.8/dev-manual/dev-manual.html#working-with-licenses
https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/#license-specification
https://rpm-packaging-guide.github.io/#preparing-source-code-for-packaging
etc.

aescolar, 2019-06-24:
> Zephyr bills itself as Apache-2.0, which I believe to be true of the main repository.

(Being a bit pedantic here, but I'd think we need to be much more precise in this discussion)
@pabigot The main repo would be Apache-2.0 _compatible_. There is BSD & MIT code in `ext/` and in some other places in-tree.
I guess that is what you meant to say? and therefore a "compatible" should be added to your question 0?

@carlescufi It may be worth rephrasing the top comment, removing the "OSS" term, and stating clearly where it is meant "Apache-2 compatible" or where it refers to any other form of open source licensed SW (if the intent was to include GPL or MPL in the discussion that would change things).

pabigot, 2019-06-24:
> > Zephyr bills itself as Apache-2.0, which I believe to be true of the main repository.
> 
> (Being a bit pedantic here, but I'd think we need to be much more precise in this discussion)
> @pabigot The main repo would be Apache-2.0 _compatible_. There is BSD-3 clause & MIT code in `ext/` and in some other places in-tree.
> I guess that is what you meant to say? and therefore a "compatible" should be added to your question 0?

No, I meant to say Apache-2.0.  The `LICENSE` file at the top says that's it's license.

The following command:

    git grep SPDX-License | grep -v Apache-2.0 | sed -e 's@^.*:@@' | acut 1| sort | uniq

shows Zephyr, as of a recent master, has these licenses in its `SPDX-License-Identifier` tags:
* BSD-3-Clause
* BSD-5-Clause-Nordic
* CC0-1.0
* GPL-2.0
* ISC

I am informed by this.

aescolar, 2019-06-24:
@pabigot ( sorry, I should have phrased my statement as "The main repo would contain Apache-2.0 compatible code" (irrespectively of the project overall being licensed as Apache2.0). It may have caused some confusion )

@pabigot Did you mean to say that you would like to restrict the main repo, and repos pulled by the default manifest, to contain _only_, Apache2.0 code?. And bar non-Apache2.0, but Apache2.0 compatible code?

(Note, AFAIK, GPL are only some development scripts used, and some of the QEMU coverage code)

pabigot, 2019-06-24:
I presented as a naive new user who assumed the `LICENSE` file gave me a correct description of the licensing that applied to the content of the tree it's in.  I had only intended to make a clear statement that preserved the existing state.

Since Zephyr does not accurately describe the licensing of its contained material (at least in the place a naive user would expect to find it), perhaps that should be addressed.  I am not a lawyer, but at least one previous employer would see those GPL scripts and say "No Zephyr on our networks", at least until somebody in upper management approved an exception, and I'd probably be in trouble if I'd gotten initial approval on the basis of claiming it to be Apache-2.0 (not "-compatible", since the company policy isn't concerned with somebody assessment of "compatible": they want a list of licenses).

jfischer-phytec-iot, 2019-06-26:
I am against it to host any binary objects or source code licensed under non-OSI approved (rather: _"popular and widely-used or with strong communities"_) licenses in the zephyrproject-rtos GitHub organization, which could be compiled and linked to an Zephyr RTOS based application (firmware). I think it is up to vendor to host it, specially from legal perspective, and the project should not be easy-peasy-barrier-free regarding code licensed under non-OSI-_"popular and widely-used or with strong communities"_ or incompatible license because of respect for those who bring and maintain their drivers mainline.

>Since Zephyr does not accurately describe the licensing of its contained material (at least in the place a naive user would expect to find it), perhaps that should be addressed.

`naive user` is subjective, it is described [here](https://docs.zephyrproject.org/latest/introduction/index.html#licensing) and [here(but outdated)](https://docs.zephyrproject.org/latest/LICENSING.html#zephyr-licensing)

carlescufi, 2022-08-19:
The TSC approved this and then a series of PRs have been bringing in support into the tree:

- #47678
- #49037
- #49194

PR #49233 (2022-08-25): blobs: Document the submission process
Add the missing piece of documentation for binary blobs: the submission
process. This follows the external source code process relatively close,
but it is relatively simpler.

The proposal includes a new issue template for requesting inclusion of
blob(s) for a particular module.

Fixes #7516.
commit 80b4df88 (2022-08-25): blobs: Document the submission process

Add the missing piece of documentation for binary blobs: the submission
process. This follows the external source code process relatively close,
but it is relatively simpler.

The proposal includes a new issue template for requesting inclusion of
blob(s) for a particular module.

Fixes #7516.
closed 2022-08-25 by nashif