zephyrproject-rtos/zephyr #7520: Feature request: Timer management API

pizi-nordic, 2018-05-14:
The generic timer API provided by the Zephyr does not abstract low-level timer properties, which are essential in low power embedded applications. As result the user is not able to fully utilize hardware capabilities.

We can easily identify four key parameters of each hardware timer in the system:
- Resolution
- Accuracy
- Power Consumption
- Start-up Time

Depending on the use case and current system state a different timer might be used as a time source.
For example a simple Bluetooth application might use 3 different timers during normal operation:
- The low-frequency RC oscillator may be used during system wakeup, as it has minimal start-up time.
- When idle, a more accurate XTAL-based oscillator could be used to reduce power consumption (Bluetooth RX window might be shortened if accurate clock source is available). 
- From time to time, when some activity is performed, a high frequency oscillator might be turned on in order to increase time base resolution.

This issue has been created as a generalization of https://github.com/zephyrproject-rtos/zephyr/issues/6498

Things needed:
- [ ]   Abstraction of low-level timer properties (resolution, accuracy, startup time, maybe power consumption)
- [ ] Universal and precise abstraction of time (time cannot be stored in timer "clocks" as time base could change at any moment).
- [ ] API for selecting most suitable clock source in run-time.
- [ ] API informing subsystems about clock source change (BT will need this to adjust RX window).

pizi-nordic, 2018-05-14:
CC: @carlescufi @cvinayak

andyross, 2018-05-14:
Noting for posterity that such a feature is likely to interact in complicated ways with tickless kernel support, which in some sense speaks to exactly the same use cases and should be unified.  Honestly tickless is a little bitrotten right now -- really we should be using it by default on most platforms where support is available.

carlescufi, 2020-03-17:
@nordic-krch could you please check if this still applies?

andyross, 2024-02-09:
Closing.  It seems like most of the practical requirements here have long since been met, we have multiple architectures that can do clock switching across idle.  We can always reopen to argue about design, but I think half a decade is enough time for this to have matured.

closed 2024-02-09 by andyross