zephyrproject-rtos/zephyr #7536: test: kernel.timer: fails on riscv32

b0661, 2018-05-15:
tests/kernel/timer/timer_api/kernel.timer fails on qemu_rscv32 on master.

qemu_riscv32              tests/kernel/timer/timer_api/kernel.timer          FAILED: failed
        see: sanity-out/qemu_riscv32/tests/kernel/timer/timer_api/kernel.timer/handler.log
```
***** Booting Zephyr OS v1.11.0-1194-g4b0b65c1b *****
Running test suite timer_api
===================================================================
starting test - test_timer_duration_period

    Assertion failed at /media/zephyr/zephyr-bobby2/tests/kernel/timer/timer_api/src/main.c:113: test_timer_duration_period: tdata
.expire_cnt == 4 is false

FAIL - test_timer_duration_period
===================================================================
starting test - test_timer_period_0

    Assertion failed at /media/zephyr/zephyr-bobby2/tests/kernel/timer/timer_api/src/main.c:130: test_timer_period_0: tdata.expire
_cnt == 1 is false

FAIL - test_timer_period_0
===================================================================
starting test - test_timer_expirefn_null
PASS - test_timer_expirefn_null
===================================================================
starting test - test_timer_periodicity
PASS - test_timer_periodicity
===================================================================
starting test - test_timer_status_get
PASS - test_timer_status_get
===================================================================
starting test - test_timer_status_get_anytime

    Assertion failed at /media/zephyr/zephyr-bobby2/tests/kernel/timer/timer_api/src/main.c:208: test_timer_status_get_anytime: k_
timer_status_get(&timer) == 4 is false

FAIL - test_timer_status_get_anytime
===================================================================
starting test - test_timer_status_sync
PASS - test_timer_status_sync
===================================================================
starting test - test_timer_k_define
PASS - test_timer_k_define
===================================================================
starting test - test_timer_user_data
PASS - test_timer_user_data
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

It also fails on PR #6761

see https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/14908/4/tests

nashif, 2018-05-17:
cant reproduce this...

b0661, 2018-05-21:
Retested  on top of  1.12.0-rc1 ->  all passed

closed 2018-05-21 by b0661