zephyrproject-rtos/zephyr #7541: arm: struct k_thread->entry is overwritten once the thread is scheduled

vaussard, 2018-05-15:
The `struct __thread_entry *entry` pointer in `struct k_thread` is useless for thread monitoring because it is located at the top of the stack and is overwritten by the first push when entering the thread.
1. `pInitCtx` is allocated at the top of the stack
2. `struct __thread_entry *entry` points to `pInitCtx`
3. When swapping the thread, `pInitCtx` is used and then popped
4. Now `struct __thread_entry *entry` points to some free stack that will be overwritten soon

This problem is easy to spot while trying to use thread->entry->pEntry to get the entry point of each thread, either in a GDB script or using some C code.

Idea to fix this issue (if CONFIG_THREAD_MONITOR):
- Reserve some space for `struct __thread_entry` at the bottom of the stack (or just after the sentinel if any)
- Copy part of `pInitCtx` (entry point + arguments) to this space and make `entry` point to it

Would it be acceptable to reserve some space at the bottom of each stack for this purpose?

nashif, 2018-05-27:
@andyross  ping

andyross, 2018-06-04:
Path up with exactly the suggested fix (which is how it's done everywhere else).  I note that we have pretty poor test coverage of CONFIG_THREAD_MONITOR though (just a few tests, none of which seem to want to enumerate threads such that they could catch this bug), so it would be helpful for the reporter to validate.

vaussard, 2018-06-05:
Thank you @andyross for fixing this issue. I will be able to test only when coming back from holidays, but this looks right. I will see if I can add a test at the same time.