zephyrproject-rtos/zephyr #7550: Generate device instance defines from dts

galak, 2018-05-15:
We should generate a define for each device instance that can be used to replace Kconfig symbols for device instance.  For example disco_l475_iot1 we have something like:

```
/* i2c@40005400 */
#define ST_STM32_I2C_V2_40005400_LABEL                  "I2C_1"

/* i2c@40005800 */
#define ST_STM32_I2C_V2_40005800_LABEL                  "I2C_2"
```

We should be able to generate defines like:

```
#define CONFIG_DTS_I2C_1 1
#define CONFIG_DTS_I2C_2 1
```

I suggest having the '_DTS_' so we don't explicitly conflict with Kconfig symbols that we can't easily replace at this time.

b0661, 2018-05-15:
@galak @erwango 

I would like to suggest a more general solution to make active device instances known. The defines should be created for the top level node the device belongs to (usually SOC but may be different). The defines should alias the device define prefix to be used to access all kind of device data.

```
/* soc */
#define SOC_GPIO_CONTROLLER_0		ST_STM32_GPIO_PINCTRL_48000000
#define SOC_GPIO_CONTROLLER_1		ST_STM32_GPIO_PINCTRL_48000400
#define SOC_GPIO_CONTROLLER_2		ST_STM32_GPIO_PINCTRL_48000800
#define SOC_GPIO_CONTROLLER_3		ST_STM32_GPIO_PINCTRL_48000C00
#define SOC_GPIO_CONTROLLER_4		ST_STM32_GPIO_PINCTRL_48001400
#define SOC_GPIO_CONTROLLER_COUNT	5
#define SOC_I2C_CONTROLLER_0		ST_STM32_I2C_V2_40005400
#define SOC_I2C_CONTROLLER_1		ST_STM32_I2C_V2_40005800
#define SOC_I2C_CONTROLLER_COUNT	2
#define SOC_PIN_CONTROLLER_0		ST_STM32_PINCTRL_48000000
#define SOC_PIN_CONTROLLER_COUNT	1
#define SOC_SPI_CONTROLLER_0		ST_STM32_SPI_FIFO_40013000
#define SOC_SPI_CONTROLLER_COUNT	1
```

To access the label of the second I2C controller you may use:

````
#ifdef SOC_I2C_CONTROLLER_1
printk(..., SOC_I2C_CONTROLLER_1##_LABEL)
#endif
````

In #7213 there was a discussion when/ how to use the CONFIG_ prefix. Here the CONFIG_xxx defines may be generated as aliases - if needed alltogether. 

This general solution is part of the RFC PR #6761 and is also supported by the inline code generation RFC PR #6762. 

You may look at https://github.com/b0661/zephyr/blob/3608375b8d54fdae0565b6c4122bb4477a935138/scripts/dts/extract/controller.py for the implementation.

Thx. Bobby

erwango, 2018-05-25:
@b0661 , thanks for head's up. We'll take it into consideration.

erwango, 2018-06-06:
Topic partly discussed here: #6762 

@b0661 proposition studied, but relies on the instantiation of all devices as a thing-controller, which is not in line with Linux compatible use of device tree.