zephyrproject-rtos/zephyr #7569: test: posix/pthread_rwlock

aescolar-ot, 2018-05-16:
tests/posix/pthread_rwlock/portability.posix calls `pthread_attr_destroy()` on the uninitialized, stack allocated, `pthread_attr_t attr[N_THR];` leading to undefined behaviour:
```
starting test - test_rw_lock

main acquire WR lock and 3 threads acquire RD lock
==18028== Thread 5:
==18028== Conditional jump or move depends on uninitialised value(s)
==18028==    at 0x80507F5: zap_pthread_attr_destroy (pthread.c:533)
==18028==    by 0x8049D4E: test_rw_lock (posix_rwlock.c:83)
==18028==    by 0x80559FC: run_test_functions (ztest.c:56)
==18028==    by 0x8055BB7: test_cb (ztest.c:196)
==18028==  Uninitialised value was created by a stack allocation
==18028==    at 0x804982F: test_rw_lock (posix_rwlock.c:55)
```

ramakrishnapallala, 2018-05-16:
I ran this on qemu_x86 and I don't see any issue. test passed. How do i reproduce this?

aescolar-ot, 2018-05-16:
compile for native_posix, and then run the executable inside valgrind
`valgrind --track-origins=yes zephyr/zephyr.exe`

aescolar-ot, 2018-05-16:
Offtopic: 
You can also use a hack like this to have `sanitycheck` run all tests in valgrind, if you prefer to run several tests with `sanitycheck`
```diff
diff --git a/scripts/sanitycheck b/scripts/sanitycheck
index f837dc0..9c49c7e 100755
--- a/scripts/sanitycheck
+++ b/scripts/sanitycheck
@@ -423,7 +423,7 @@ class NativeHandler(Handler):
         """
         super().__init__(instance)
 
-        self.valgrind = False
+        self.valgrind = True
 
     def _output_reader(self, proc, harness):
         log_out_fp = open(self.handler_log, "wt")
@@ -450,7 +450,8 @@ class NativeHandler(Handler):
         command = [binary]
         if shutil.which("valgrind") and self.valgrind:
             command = ["valgrind", "--error-exitcode=2",
-                       "--leak-check=full"] + command
+                       "--leak-check=full", "--log-fd=1", "--track-origins=yes"  ] + command
+            self.timeout = self.timeout*10
 
         with subprocess.Popen(command, stdout=subprocess.PIPE, stderr=subprocess.PIPE) as proc:
             t = threading.Thread(target=self._output_reader, args=(proc, harness, ))
@@ -463,8 +464,10 @@ class NativeHandler(Handler):
 
             proc.wait()
             self.returncode = proc.returncode
-            if proc.returncode != 0:
+            if self.returncode != 0:
                 out_state = "failed"
+            else:
+                out_state = "passed"
 
         returncode = subprocess.call(["GCOV_PREFIX=" + self.outdir, "gcov", self.sourcedir, "-b", "-s", self.outdir], shell=True)
 
@@ -475,6 +478,9 @@ class NativeHandler(Handler):
         else:
             self.set_state(out_state, {})
 
+        if out_state == "failed":
+            self.set_state(out_state, {})
+
 
 class UnitHandler(Handler):
     def __init__(self, instance):
```

aescolar-ot, 2018-05-18:
@ramakrishnapallala Are you able to reproduce it?

ramakrishnapallala, 2018-05-18:
@aescolar-ot I was looking into some other stuff, will look into it shortly.

ramakrishnapallala, 2018-05-18:
@aescolar-ot I have looked at the code. From the pthread_attr_destroy() I don't see any issue as it returns 0 or EINVAL.

From the test case point, we can set the attr[n_THR] memory to 0x0 or make it static.

Do you see any other issue?

aescolar-ot, 2018-05-18:
Why is the test calling `pthread_attr_destroy()` before calling `pthread_attr_init()`?

ramakrishnapallala, 2018-05-18:
I think the author of the test wanted to make sure the resources are freed (if allocated earlier) before he uses the attr struct for creating a new pthread.

aescolar-ot, 2018-05-18:
As it is now it is being called over uninitialized memory, 
sometimes attr->initialized is 0, sometimes it is 1, causing random coverage differences.
I think that call should be removed.

ramakrishnapallala, 2018-05-18:
Ok will fix it.

PR #7682 (2018-05-22): tests: posix: Fix pthread attribute initialization
Fix pthread_rwlock test pthread attributes initialization.

Fixes #7569
closed 2018-05-22 by nashif