zephyrproject-rtos/zephyr #7570: usb: update bcdUSB to 2.00

jfischer-phytec-iot, 2018-05-16:
There is suggestion/need to increase bcdUSB to 2.00. Are all supported controllers USB Rev. 2.0 compliant?

- Kinetis (USBFSOTG) is USB 2.0 full-speed/low-speed compliant
- NRF52 is Fully compliant to Universal Serial Bus Specification Revision 2.0 Full Speed part
- STM32 Cube HAL, are all supported controllers USB 2.0 compliant?
- usb_dc_dw.c ?
- usb_dc_sam0.c ?

@ydamigos, @carlescufi, @nzmichaelh @finikorg @loicpoulain

carlescufi, 2018-05-16:
CC @pawelzadrozniak, will defer to him

pawelzadrozniak, 2018-05-16:
There could be a configuration option to choose between 1.1 and 2.0.

jfischer-phytec-iot, 2018-05-16:
>There could be a configuration option to choose between 1.1 and 2.0.

Only if there is no other way. e.g. if we have a non 2.0 compatible controller

nzmichaelh, 2018-05-22:
The SAM0 is USB2 so you're good there.  I can't think of any reasons to support USB1.1.

PR #7822 (2018-06-11): subsys: usb: update bcdUSB to 2.00
Update bcdUSB to 2.00

resolves #7570
commit 1ce6a6ea (2018-06-11): subsys: usb: update bcdUSB to 2.00

Update bcdUSB to 2.00

resolves #7570
closed 2018-06-11 by nashif