zephyrproject-rtos/zephyr #7587: Migrate DNS to BSD sockets

laperie, 2018-05-16:
As a part of the Zephyr Networking API rework we need to update all protocols to be based on socket.h instead of net_app/net_context.

pfalcon, 2018-09-06:
As I wrote in https://github.com/zephyrproject-rtos/zephyr/issues/7591#issuecomment-394791879 , we should be fine with DNS being implemented using native API - for quite some time at least. So, this should be the least priority of all "port to sockets" items in https://github.com/zephyrproject-rtos/zephyr/issues/7591 (well, unless someone has a usecase to re-prioritize it).

pfalcon, 2018-11-29:
@laperie: Do you agree with assessment above? Maybe we should close this to better track scope of other tasks?

pfalcon, 2019-01-24:
@agross-linaro 

> This causes issues with users who assume it is synchronous and are passing in stack based arguments.

I'm sorry, but why would someone assume that? And ungrounded assumptions can be made about any API. We have docs and expect that programmers do read them as a solution to that.

>  I'd rather see people be migrated to a synchronous DNS socket API and deprecate the old stuff.

"Old stuff" is asynchronous, and that's its benefit. In my list, there's nothing to deprecate. Majority of people will simply use POSIX getaddrinfo(), but those who'll need async solution, will use the native Zephyr API. (There's no well established standard API for async DNS (to the best of my knowledge), so Z's API is as good as any other).

pfalcon, 2019-01-24:
As 1.14 release is quite close, let me nominate this ticket for dev review for possible closure. @jukkar, if you have any comments, please share.

agross-linaro, 2019-01-24:
@pfalcon Because our shell does it wrong for one.  Everyone who uses it is doing it wrong, aside from the tests and the resolve_name API.  The only documentation I see is in the include file.

pfalcon, 2019-01-24:
@agross-linaro : Well, let's fix it. If you can provide any other details, please do (perhaps, in separate ticket). If not, I'll look into it when I have a chance.

agross-linaro, 2019-01-24:
@pfalcon oh, i'm already fixing it.  I just find things like this that have other APIs that wrap them to make them synchronous to be kind of silly... and subject to misuse.

pabigot, 2019-01-31:
@pfalcon will you call in to the dev-review telecon?  If not, exactly what do you want us to discuss about this?

I'm inferring your position to be "it works fine as-is, close as not-worth-fixing".  On the other hand, this is an issue not a PR, so what's the need to resolve it for 1.14?  #7592 appears to be less than half way to completion; will leaving this be the only thing blocking closing that issue in 1.14?

My personal assessment is that this has the highest ratio of contention-to-value of the issues pending for discussion, so it should be scheduled last and we probably won't get to it.

pabigot, 2019-01-31:
Per dev-review: to be discussed in networking forum next week(?).

pfalcon, 2019-02-06:
At the networking forum yesterday, we agreed to close this. Summing up, the current situation is that:

1. We offer standard POSIX getaddrinfo() function for blocking DNS queries. This is expected to satisfy needs of the most users.
2. We offer native Zephyr DNS API, for advanced cases like async DNS resolution: https://docs.zephyrproject.org/latest/reference/networking/dns_resolve.html .

The docs on the latter definitely can, and should be improved - which is a separate matter, reports on specific issues and patches are welcome (and definitely in queue for networking team).

Otherwise, closing this, to let us concentrate on more important areas.

closed 2019-02-06 by pfalcon