zephyrproject-rtos/zephyr #7598: move CONFIG_INIT_STACKS logic for IRQ stack to common code

andrewboie, 2018-05-16:
On all arches this is currently done in arch/, usually in some early boot assembly code.
However we don't enable interrupts until we are well into the C domain, and it should be possible to do this somewhere in kernel/init.c, assuming we can standardize the symbol name and kconfig define for the interrupt stack and its size.

ioannisg, 2019-06-23:
At least for ARM, the interrupt stack is used until we initialize the kernel and jump to main. So it is not trivial to do this in init.

andrewboie, 2020-08-11:
I have a patch for this that needs a bit more work but this should be feasible for 2.4 https://github.com/zephyrproject-rtos/zephyr/pull/23533

closed 2024-02-09 by andyross