zephyrproject-rtos/zephyr #7610: tests/lib/c_lib fails on native_posix on FC28

andrewboie, 2018-05-17:
native_posix builds with the host compiler, and on Fedora 28 the tests/lib/c_lib fails sanitycheck as follows:

```
/home/apboie/projects/zephyr/tests/lib/c_lib/src/main.c: In function ‘test_strncpy’:
/home/apboie/projects/zephyr/tests/lib/c_lib/src/main.c:196:2: error: ‘strncpy’ output truncated copying 10 bytes from a string of length 26 [-Werror=stringop-truncation]
  strncpy(buffer, "This is over 10 characters", BUFSIZE);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

aescolar-ot, 2018-05-18:
Too smart compiler :)
Anyhow, I guess we could add an ignore pragma:
```diff
+#   pragma GCC diagnostic push 
+#   pragma GCC diagnostic ignored "-Wstringop-truncation"
strncpy(buffer, "This is over 10 characters", BUFSIZE);
/* Purposely different values */
ret = strncmp(buffer, "This is over 20 characters", BUFSIZE);
+#   pragma GCC diagnostic pop
```
or?

andrewboie, 2018-05-24:
@andyross fixed this

closed 2018-05-24 by andrewboie

aescolar-ot, 2018-05-24:
@andrewboie : do you have a link to the commit which fixed it?