zephyrproject-rtos/zephyr #7613: an issue about OTA/mcumgr

Jeepgoing, 2018-05-17:
hi all:
i am now testing a demo about OTA fuction, and meet some issues, my target boards is two Nordic nrf52_pca10040, one run SMP Server Sample[https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/mgmt/mcumgr/smp_svr](url), and another run Bluetooth: HCI UART [https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/hci_uart](url) .

i am true that the SMP Server Sample runs will, when it is running, it is printk some debug infomations as below:

***** Booting Zephyr OS 1.11.99 *****
[MCUBOOT] [INF] main: Starting bootloader
[MCUBOOT] [INF] boot_status_source: Image 0: magic=unset, copy_done=0xff, image_ok=0xff
[MCUBOOT] [INF] boot_status_source: Scratch: ***** Booting Zephyr OS 1.11.99 *****
[MCUBOOT] [INF] main: Starting bootloader
[MCUBOOT] [INF] boot_status_source: Image 0: magic=unset, copy_done=0xff, image_ok=0xff
[MCUBOOT] [INF] boot_status_source: Scratch: magic=unset, copy_done=0x0, image_ok=0xff
[MCUBOOT] [INF] boot_status_source: Boot source: slot 0
[MCUBOOT] [INF] boot_swap_type: Swap type: none
[MCUBOOT] [INF] main: Bootloader chainload address offset: 0xc000
[MCUBOOT] [INF] main: Jumping to the first image slot
***** Booting Zephyr OS 1.11.99 *****
Zephyr Shell, Zephyr version: 1.11.99
Type 'help' for a list of available commands
shell> Bluetooth initialized
Advertising successfully started

now the problem is i have install all things follow the guides:
[https://github.com/zephyrproject-rtos/zephyr/tree/master/ext/lib/mgmt/mcumgr](url)
[https://github.com/apache/mynewt-mcumgr-cli/blob/master/README.md](url)
[https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/mgmt/mcumgr/smp_svr](url)
[https://devzone.nordicsemi.com/b/blog/posts/nrf5x-support-within-the-zephyr-project-rtos](url)

but when i run:
sudo mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello

sometimes the target can reply hello, but more times the target reply nothing, and the terminal will be blocked, and I had to kill the terminal thread.

[image]

so who can help me or anyone else has used OTA.

carlescufi, 2018-05-18:
@Jeepgoing I will take a look to see if I can reproduce and get back to you

carlescufi, 2018-05-18:
@Jeepgoing I just tested this with vanilla Zephyr master, using a Virtual Machine just like you, but VMWare instead of VirtualBox. I used the following hardware:

* nRF51822 dongle for the `hci_uart` controller
* nRF52832 Devkit for the `smp_svr` board

I ran this 10 times with 10/10 connections:

```
carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
[sudo] password for carles:
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello

carles@zephyr ~/src/zephyr (master)
$ sudo ~/bin/go/bin/mcumgr --conntype ble --connstring ctlr_name=hci0,peer_name='Zephyr' echo hello
hello
```

Could you please tell me which hardware exactly you are using in order to try it in the same conditions as you are?

Jeepgoing, 2018-05-21:
thank you for your answer, and i will do more test.@carlescufi

carlescufi, 2018-05-28:
@Jeepgoing can you please let us know if this is still an issue or if we can close it? Thanks!

Jeepgoing, 2018-05-28:
sorry, i haven't done this lately, but i think you are right, this issue maybe caused by my development environment, and when i have time, i will do more test, now i will close this issue.

closed 2018-05-28 by Jeepgoing