zephyrproject-rtos/zephyr #7654: BLE GATT notification: no information about the state of the notifications queue

pdunaj, 2018-05-18:
I can schedule notifications and they are queued up to some point. When queue is full data is lost but I am not notified about it. Also as there is no callback informing notification was sent, I cannot tell if there is place on the queue to put a new notification.

Is there a way to get information about state of the queue where notifications are stored, or even better get a callback when notification is being send?

Vudentz, 2018-05-22:
There is no such thing as a queue limit, the TX will keep queueing forever, what it should happen instead is if there are no buffer available which means bt_gatt_notify can block until a buffer becomes available. Obviously, if a thread is blocked it won't be able to queue more data, in which case I would see a point of having a timeout, which is something that perhaps applies to more than bt_gatt_notify so the thread can give up or retry if the notification is not sent within that time.

Btw, I don't think HCI provides an event to when a specific packet is sent, we assume that is on Packet Complete but afaik that doesn't guarantee it was sent over the air, anyway for a transaction that actually has a response that perhaps is not needed and those that don't need response do normally block so waiting for a complete event maybe tricky because the callback might be called from a different context, possible RX thread.

carlescufi, 2018-05-23:
@pdunaj is this a duplicate of #7659 ?

carlescufi, 2018-05-23:
>I don't think HCI provides an event to when a specific packet is sent, we assume that is on Packet Complete but afaik that doesn't guarantee it was sent over the air

Well, the spec does say that it has either been sent over the air or flushed (which is not possible for BLE):

The Number Of Completed Packets event is used by the Controller to indicate
to the Host how many HCI Data Packets have been completed (transmitted or
flushed) for each Connection_Handle since the previous Number Of
Completed Packets event was sent to the Host.

closed 2018-11-13 by Vudentz