zephyrproject-rtos/zephyr #7657: Invalid argument passed to k_sleep

pdunaj, 2018-05-18:
https://github.com/zephyrproject-rtos/zephyr/blob/0b76b12837deb995077bf026de01a6e9a9e50dfe/tests/kernel/semaphore/semaphore/src/main.c#L202

Probably confised with K_MSEC.

Present in multiple places.
Check also:
https://github.com/zephyrproject-rtos/zephyr/blob/0b76b12837deb995077bf026de01a6e9a9e50dfe/tests/net/traffic_class/src/main.c#L516

https://github.com/zephyrproject-rtos/zephyr/blob/0b76b12837deb995077bf026de01a6e9a9e50dfe/subsys/net/lib/app/net_app.c#L1300

jukkar, 2018-05-22:
The @7711 is fixing the networking part of this issue.

pfalcon, 2018-05-23:
#7711 fixes only subset of the issues listed here, so reopening.

I also submitted https://github.com/zephyrproject-rtos/zephyr/issues/7817 arguing that the root cause of these issues is that SECONDS() and MSEC() are named confusingly for what they do.

PR #7711 (2018-05-22): net: Too long timeout was passed to k_sleep
The code was using MSEC() macro in few places instead of more
proper K_MSEC(). The MSEC() takes seconds as a parameter and
K_MSEC() takes milliseconds.

Fixes #7657
commit 6180b904 (2018-05-22): net: Too long timeout was passed to k_sleep

The code was using MSEC() macro in few places instead of more
proper K_MSEC(). The MSEC() takes seconds as a parameter and
K_MSEC() takes milliseconds.

Fixes #7657
PR #7849 (2018-05-28): Fix k_sleep() timeouts in net
This converts remaining MSEC() calls to K_MSEC() in networking code.
Also convert MSEC_PER_SEC to K_SECONDS() as that is more intuitive.

Fixes #7657
closed 2018-05-28 by nashif