zephyrproject-rtos/zephyr #7658: [RFC] net: Split off net_app_settings lib to a separate directory under subsys/net/lib/

pfalcon, 2018-05-18:
Over last time, there was a lot of talk that "net_app API does too much things at once", which leads to confusion. At the same time, net_app is (at least) 2 largely independent parts:

1. Client/server helper API.
2. Network setting module.

It should be noted that p.2 isn't even an "API", it's a module, or subsystem, which is driven by "declarative" Kconfig settings and just configures network environment on the system startup.

To make this distinction clear, I propose to move all file related to settings module from subsys/net/lib/app/ to subsys/net/lib/settings/. This would affect init.c, bt_settings.[ch], ieee802154_settings.[ch].

To make the move clean, we may also want to rename Kconfig options, e.g. NET_APP_AUTO_INIT => NET_SETTINGS_AUTO_INIT. (Or alternatively, we can use subsys/net/lib/config/ and NET_CONFIG_*).

pfalcon, 2018-05-18:
@jukkar, @tbursztyka : Let me know if this makes sense, I can prepare a patch for 1.13.

@mike-scott, @mbolivar, @d3zd3z : FYI

jukkar, 2018-05-18:
+1 from me.

I just wonder if the _settings_ could cause confusion with the settings API, so in this respect the _config_ might be better choice.

d3zd3z, 2018-05-22:
I think this is a really good start. Regardless of how the networking code is used, it will have to be configured somehow.

My concern with this setup is that it seems overly tailored to testing, and isn't particularly useful for an application using Zephyr. It is reasonable that a testing environment uses a hard-coded network configuration. However, this doesn't really make sense for an actual application. In this instance, it is going to have to be one or more of:

1. Have a static configuration, but stored in some kind of configuration area. Maybe there is a UI or a web interface that allows the user to set this configuration.
2. DHCP. The system will need to try to get network configuration from dhcp (and any networking part of the app will need to wait for it). This is currently mixed into one of the many things that net_app is doing.

pfalcon, 2018-06-04:
@jukkar:

> I just wonder if the settings could cause confusion with the settings API, so in this respect the config might be better choice.

I assume you mean "settings in the non-volatile storage" API, CONFIG_SETTINGS. Using "config" sounds good then.

pfalcon, 2018-06-04:
@d3zd3z:

> My concern with this setup is that it seems overly tailored to testing, and isn't particularly useful for an application using Zephyr.

I can definitely ack that, but I don't see that as a concern. It all boils down to whether we work on a generic RTOS, or some shrink-wrapped functionality toolset. In my list, it's the first - at least, first of all. And then testing is very important part of it. Doing stuff for real-world production applications is important too, but there're supposedly people who concentrate on that, and such a support may be as well specific for each application/product (as an example, you talk about DHCPv4, but we were told that IPv6 will rule the world, and there's no need (or support) for DHCPv4 in it). I'd be happy to contribute to that part of development too, but hopefully, after the bugs in the IP stack are fixed, as that makes developing real-world apps hard at all.

All in all, it's interesting topic to discuss, but I guess it's outside the scope of this ticket.

mike-scott, 2018-06-04:
+1 from here.  Especially with the move from net_app APIs to socket APIs scheduled for 1.13/1.14, I think the timing is right to move out the net config/settings.

PR #8460 (2018-06-20): net: app: Split code for configuring network to a separate lib, "config"
net_app library consisted of 2 disjoint parts: a) library to
setup/configure networking on the application startup; b) library
to setup client and server connections.

As the configuration library is universally useful and is a generic
Zephyr networking component, split it out to a seperate top-level
networking library under net/lib/config/.

Fixes: #7658
commit c60df131 (2018-06-20): net: app: Split code for configuring network to a separate lib, "config"

net_app library consisted of 2 disjoint parts: a) library to
setup/configure networking on the application startup; b) library
to setup client and server connections.

As the configuration library is universally useful and is a generic
Zephyr networking component, split it out to a seperate top-level
networking library under net/lib/config/.

Fixes: #7658
closed 2018-06-20 by jukkar