zephyrproject-rtos/zephyr #7671: NVS broken for write-align > 4

aurel32, 2018-05-19:
NVS has support for handling write alignment, by "padding" the header and data up to the write alignment.

However the implementation has some issues:
- When writing a header or a footer, some garbage might be written to the flash:
```
struct _nvs_data_hdr data_hdr;
...
hdr_len = _nvs_len_in_flash(fs, sizeof(struct _nvs_data_hdr));
...
rc = nvs_flash_write(fs, fs->write_location, &data_hdr, hdr_len);
```
`data_hdr` takes 4 bytes in memory, but `write-align` bytes are written to the flash. The remaining bytes are taken from the stack.
- The same issue happens when reading the header or a footer, except that this time it causes stack corruption and thus non-predictable behavior:
```
hdr_addr = _nvs_head_addr_in_flash(fs, entry);
hdr_len = _nvs_len_in_flash(fs, sizeof(struct _nvs_data_hdr));
rc = nvs_flash_read(fs, hdr_addr, &head, hdr_len);
```
- On the other hand no padding is done when writing the data. This means that for a 4-byte value, when `write-align` equals to 8 bytes, only 4 bytes ends-up being passed to flash_write and the write get rejected by the flash layer.

A quick workaround for the two first issues is to pad the structures in `nvs_priv.h`. For example with `write-align` set to 8:
```
--- a/subsys/fs/nvs/nvs_priv.h
+++ b/subsys/fs/nvs/nvs_priv.h
@@ -47,11 +47,13 @@ struct _nvs_sector_hdr {
 struct _nvs_data_hdr {
        u16_t id;
        u16_t len;
+       u32_t _pad;
 };
 
 struct _nvs_data_slt {
        u16_t crc16;
        u16_t _pad;
+       u32_t _pad2;
 };
```

carlescufi, 2018-05-19:
@Laczen can you take a look please?

Laczen, 2018-05-19:
@aurel32, thank you for reporting this and the proposed solution. Your analysis is correct. I will work on a solution that is more general as the same problem will appear with the sector_hdr if ''write-align'' is larger than 8 bytes.

Laczen, 2018-05-19:
@carlescufi A fix has been proposed in PR #7600.

aurel32, 2018-05-19:
@Laczen, I have tried your patch, it indeed fixes the two first point, but I believed the fix is very fragile and that the bug can reappear with a different compiler or in case of small code change.

The third point, ie writing data smaller than the block size still doesn't work.

aurel32, 2018-05-19:
@Laczen, I don't thinks it's a good idea to set the write alignment to a different value than the block size, even if the SoC allows writing smaller than the block size. As you said what is important to guarantee the integrity, is to write the header, the data and the slot in different blocks. This can only be safely guaranteed by correctly defining the block size.

Then you can even get rid of the `_pad` entries in `_nvs_sector_hdr` and `_nvs_data_slt`, which seems there only to guarantee that when block size equals to 4, and which just waste space when the block size equals 1 or 2.

PR #7674 (2018-05-26): subsys: fs/nvs: fix write block size issues
This patch series is an alternative to #7600 to fix write block size issues. It ensures that all writes to the flash are done in multiple of the write block size, even if the data to write is smaller, by adding some padding. It also make sure that the flash read and write match the destination or source buffer to avoid stack corruption or data leakage.

This fixes #7671 for me, running on a nucleo_l432kc board (8-byte write block, smaller writes not possible).

@Laczen, note that it only fixes the write block size issues, I guess there are more thing [... cut]
closed 2018-05-26 by nashif