zephyrproject-rtos/zephyr #7676: buildsystem: 'make flash' failed on nucleo_f070rb platform

zhaozhongchen, 2018-05-21:
work tree: 1.12.0-rc1
next is the error log:

```
Scanning dependencies of target flash
[100%] Flashing nucleo_f070rb
fatal error: --zephyr-base missing and no ZEPHYR_BASE in the environment
usage: west [-h] [-z ZEPHYR_BASE] [-v] {flash,debug,debugserver} ...
zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:57: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash' failed
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash] Error 1
CMakeFiles/Makefile2:1984: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash.dir/all' failed
make[2]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/all] Error 2
CMakeFiles/Makefile2:1991: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash.dir/rule' failed
make[1]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Error 2
Makefile:391: recipe for target 'flash' failed
make: *** [flash] Error 2
```

zhaozhongchen, 2018-05-21:
env ZEPHYR_BASE was defined when using 'make flash', but can't find it in os.environ when using 'sudo make flash'.

```
        if 'ZEPHYR_BASE' not in os.environ:
            raise InvalidWestContext(
                '--zephyr-base missing and no ZEPHYR_BASE '
                'in the environment')
        else:
            args.zephyr_base = os.environ['ZEPHYR_BASE']
```

chunshanzhu, 2018-05-21:
sudo use the root environment. use 'make flash'

zhaozhongchen, 2018-05-21:
@chunshanzhu but 'make flash' can't access usb because doesn't has authority.

```
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: open failed
```

mbolivar, 2018-05-21:
@zhaozhongchen thank you for reporting this issue.

> @chunshanzhu but 'make flash' can't access usb because doesn't has authority.

The flash command was recently changed to a new program which checks for ZEPHYR_BASE. I have sent a patch which removes this check, but for your information, it is not recommended to run the flash command as root.

On Linux using OpenOCD (your board uses OpenOCD to flash and debug), the recommended way to get privileges is to install the following file into your /etc/udev/rules.d directory:

https://github.com/zephyrproject-rtos/openocd/blob/master/contrib/60-openocd.rules

Make sure you are a member of the plugdev group, and that you unplug and plug back in the board so the new rules take effect.

@erwango, I notice that the ST board documentation doesn't make note of any required udev rules. Is this an oversight, or are you recommending users run as root?

zhaozhongchen, 2018-05-21:
@mbolivar thanks for yours resolution, it's really helpful to me

PR #7688 (2018-05-22): scripts: west: downgrade missing ZEPHYR_BASE to a warning
This is not strictly necessary for flashing and debugging, and is
causing issues in Linux environments where users run "make flash" as
root instead of installing udev rules.

Fixes: #7676

(Though we will need to revisit this when adding commands that run
CMake).
commit e0477756 (2018-05-22): scripts: west: downgrade missing ZEPHYR_BASE to a warning

This is not strictly necessary for flashing and debugging, and is
causing issues in Linux environments where users run "make flash" as
root instead of installing udev rules.

Fixes: #7676

(Though we will need to revisit this when adding commands that run
CMake).
closed 2018-05-22 by nashif