zephyrproject-rtos/zephyr #7677: mcuboot-master imgtool.py sign error

changkong5, 2018-05-21:
lsb@lsb-ThinkPad-E470:zephyr-master-v1.11.99.x$ ./mcuboot-master/scripts/imgtool.py sign \
>      --key ./mcuboot-master/root-rsa-2048.pem \
>      --header-size 0x200 \
>      --align 8 \
>      --version 1.0 \
>      --included-header \
>      ./usr/nrf52/smp_svr/build/zephyr/zephyr.hex signed.hex
Traceback (most recent call last):
  File "./mcuboot-master/scripts/imgtool.py", line 150, in <module>
    args()
  File "./mcuboot-master/scripts/imgtool.py", line 147, in args
    subcmds[args.subcmd](args)
  File "./mcuboot-master/scripts/imgtool.py", line 85, in do_sign
    img.sign(key)
  File "/home/lsb/Desktop/zephyr-master-v1.11.99.x/mcuboot-master/scripts/imgtool/image.py", line 135, in sign
    sig = key.sign(bytes(self.payload))
  File "/home/lsb/Desktop/zephyr-master-v1.11.99.x/mcuboot-master/scripts/imgtool/keys/rsa.py", line 92, in sign
    return self.key.sign(
AttributeError: '_RSAPrivateKey' object has no attribute 'sign'
lsb@lsb-ThinkPad-E470:zephyr-master-v1.11.99.x$

Jeepgoing, 2018-05-21:
look at this issue, maybe can give you some advise.[https://github.com/runtimeco/mcuboot/issues/253](url)

changkong5, 2018-05-21:
The problem was solved. thanks for you help.

closed 2018-05-21 by changkong5