zephyrproject-rtos/zephyr #7692: Kernel tests failing at runtime on frdm_k64f

MaureenHelm, 2018-05-21:
Several kernel tests are failing at runtime on frdm_k64f on commit e7509c17c5938b48f9ad4b7ae78199bb0faa6113 (v1.12.0-rc1)
```
$ sanitycheck -T tests/kernel/ -p frdm_k64f --device-testing --device-serial /dev/ttyACM0
Cleaning output directory /home/maureen/zephyr/sanity-out
Building testcase defconfigs...
72 tests selected, 8595 tests discarded due to filters
total complete:    7/  72   9%  failed:    0

frdm_k64f                 tests/kernel/stack/stack_api/kernel.stack          FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/stack/stack_api/kernel.stack/handler.log

total complete:   19/  72  26%  failed:    1

frdm_k64f                 tests/kernel/msgq/msgq_api/kernel.message_queue    FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/msgq/msgq_api/kernel.message_queue/handler.log

total complete:   22/  72  30%  failed:    2

frdm_k64f                 tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/handler.log

total complete:   39/  72  54%  failed:    3

frdm_k64f                 tests/kernel/queue/kernel.queue.poll               FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/queue/kernel.queue.poll/handler.log

total complete:   45/  72  62%  failed:    4

frdm_k64f                 tests/kernel/mem_protect/mem_protect/kernel.memory_protection FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/mem_protect/mem_protect/kernel.memory_protection/handler.log

total complete:   57/  72  79%  failed:    5

frdm_k64f                 tests/kernel/queue/kernel.queue                    FAILED: timeout
	see: sanity-out/frdm_k64f/tests/kernel/queue/kernel.queue/handler.log

total complete:   72/  72  100%  failed:    6
66 of 72 tests passed with 0 warnings in 1570 seconds
```

Most are failing with a bus fault:
```
***** Booting Zephyr OS v1.12.0-rc1 *****                                                      
Running test suite stack_api
===================================================================
starting test - test_stack_thread2thread
***** BUS FAULT *****
  Executing thread ID (thread): 0x20000350
  Faulting instruction address:  0x900
  Precise data bus error
  Address: 0x2000217c
Fatal fault in thread 0x20000350! Aborting.
```

MaureenHelm, 2018-05-21:
@andrewboie I bisected the first failure `tests/kernel/stack/stack_api` to the following commit:
```
f3bee951b1d5944492db70e26c3e03cd1be1c645 is the first bad commit
commit f3bee951b1d5944492db70e26c3e03cd1be1c645
Author: Andrew Boie <andrew.p.boie@intel.com>
Date:   Wed May 2 17:44:39 2018 -0700

    kernel: stacks: add k_stack_alloc() init
    
    Similar to what has been done with pipes and message queues,
    user mode can't be trusted to provide a buffer for the kernel
    to use. Remove k_stack_init() as a syscall and offer
    k_stack_alloc_init() which allocates a buffer from the caller's
    resource pool.
    
    Fixes #7285
    
    Signed-off-by: Andrew Boie <andrew.p.boie@intel.com>
```

andrewboie, 2018-05-21:
I'm trying to look into it. Getting this frdm board to flash is fighting me every step of the way, we could really use a documentation pass on http://docs.zephyrproject.org/boards/arm/frdm_k64f/doc/frdm_k64f.html it is confusing for someone not familiar with the development ecosystem

andrewboie, 2018-05-21:
stack_api was failing due to CONFIG_USERSPACE being turned on but not CONFIG_APPLICATION_MEMORY. didn't fail in qemu_x86 due to CONFIG_APPLICATION_MEMORY being defaulted on. Looking at the right way to enable user mode for certain tests on boards where user mode isn't in the default configuration, I think @AdithyaBaglody had proposed something.

andrewboie, 2018-05-21:
I submitted a patch that corrected this, as well as enabling test cases that run under user mode to enable user mode if the hardware supports it.
Unfortunately, doing that has exposed a host of issues which I am currently working through.
The most vexing is #7704 I need @agross-linaro help on this.

andrewboie, 2018-05-22:
I have fixed or applied workarounds (with bugs filed) for everything except a problem with one test which is so far mysterious and seems to be some kind of data/stack corruption:

```
-----sanity-out/frdm_k64f/tests/kernel/threads/lifecycle/thread_init/kernel.threads/handler.log-----
***** Booting Zephyr OS v1.12.0-rc1-11-gfe71779cb *****
Running test suite thread_init
===================================================================
starting test - test_kdefine_preempt_thread
0x00007e1a is not a valid sem
syscall _handler_k_sem_give failed check: access denied
**** Unknown Fatal Error 0! ****
Current thread ID = 0x200028c4
Faulting instruction address = 0x9fd
Fatal fault in thread 0x200028c4! Aborting.
***** BUS FAULT *****
  Executing thread ID (thread): 0x20002838
  Faulting instruction address:  0x0
  Stacking error
Fatal fault in thread 0x20002838! Aborting.
***** BUS FAULT *****
  Executing thread ID (thread): 0x20002694
  Faulting instruction address:  0x0
  Stacking error
Fatal fault in thread 0x20002694! Aborting.
-----sanity-out/frdm_k64f/tests/kernel/threads/lifecycle/thread_init/kernel.threads/handler.log-----
```

andrewboie, 2018-05-22:
#7752 should fix this.
Another issue has been found by enabling more tests, tracked in #7754

closed 2018-05-22 by andrewboie

pfalcon, 2018-05-23:
@andrewboie :

> Getting this frdm board to flash is fighting me every step of the way, we could really use a documentation pass on http://docs.zephyrproject.org/boards/arm/frdm_k64f/doc/frdm_k64f.html it is confusing for someone not familiar with the development ecosystem

To "flash" this board (and bunch others, arm mbed compatible), you just need to connect it over usb, wait for a flash disk to mount and run:

~~~
cp outdir/frdm_k64f/zephyr/zephyr.bin /media/pfalcon/MBED/
~~~

The docs are indeed a bit over-engineered...

commit bdc2e84f (2018-05-21): tests: enable user mode by default

Many boards do not enable user mode in their default configuration
even though they support it. However for purposes of our unit
testing we really do want it on all the time.

It's not enough to just flip on CONFIG_USERSPACE, the tests largely
assume that CONFIG_APPLICATION_MEMORY and CONFIG_DYNAMIC_OBJECTS
are enabled. Update CONFIG_TEST_USERSPACE to be on by default if
CONFIG_TEST is active, and hang CONFIG_DYNAMIC_OBJECTS on it too.

This problem was not previously detected as qemu_x86 has the
user mode options in its defconfig and it is currently the onl [... cut]
commit efe0c4b7 (2018-05-22): tests: enable user mode by default

Many boards do not enable user mode in their default configuration
even though they support it. However for purposes of our unit
testing we really do want it on all the time.

It's not enough to just flip on CONFIG_USERSPACE, the tests largely
assume that CONFIG_APPLICATION_MEMORY and CONFIG_DYNAMIC_OBJECTS
are enabled. Update CONFIG_TEST_USERSPACE to be on by default if
CONFIG_TEST is active, and hang CONFIG_DYNAMIC_OBJECTS on it too.

This problem was not previously detected as qemu_x86 has the
user mode options in its defconfig and it is currently the onl [... cut]