zephyrproject-rtos/zephyr #7698: Kernel tests failing at runtime on frdm_kw41z

MaureenHelm, 2018-05-21:
Several kernel tests are failing at runtime on frdm_kw41z on commit e7509c1 (v1.12.0-rc1)
```
$ sanitycheck -T tests/kernel/ -p frdm_kw41z --device-testing --device-serial /dev/ttyACM0
Cleaning output directory /home/maureen/zephyr/sanity-out
Building testcase defconfigs...
63 tests selected, 8604 tests discarded due to filters
total complete:   26/  63  41%  failed:    0

frdm_kw41z                tests/kernel/fatal/kernel.common.stack_sentinel    FAILED: failed
	see: sanity-out/frdm_kw41z/tests/kernel/fatal/kernel.common.stack_sentinel/handler.log

total complete:   32/  63  50%  failed:    1

frdm_kw41z                tests/kernel/mem_protect/stackprot/kernel.memory_protection FAILED: timeout
	see: sanity-out/frdm_kw41z/tests/kernel/mem_protect/stackprot/kernel.memory_protection/handler.log

total complete:   36/  63  57%  failed:    2

frdm_kw41z                tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random FAILED: timeout
	see: sanity-out/frdm_kw41z/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/handler.log

total complete:   63/  63  100%  failed:    3
60 of 63 tests passed with 0 warnings in 896 seconds
```

andrewboie, 2018-05-21:
probably same root cause as #7692

andrewboie, 2018-05-22:
This should be fixed now by Leandro's patch #7767

nashif, 2018-05-23:
tests/kernel/mem_protect/stack_random still failing on this board:

```
**** HARD FAULT *****
  Executing thread ID (thread): 0x00000000
  Faulting instruction address:  0x1a18
Fatal fault in ISR! Spinning...
```

lpereira, 2018-05-23:
This is indeed caused by semaphores being used too early in the boot process.  A way to fix this is to use the entropy driver directly if the system has one, and fall back to sys_rand32_get() if it doesn't.  Let me cook up a patch here.

lpereira, 2018-05-23:
PR #7835 should alleviate these symptoms, but entropy drivers might use synchronization primitives early in the boot process, causing the same issue.  Commit message has the details.

andrewboie, 2018-05-23:
I think this will do it https://github.com/zephyrproject-rtos/zephyr/pull/7838

andrewboie, 2018-05-23:
Confirmed by Anas with #7838 

```
(3:31:44 PM) Anas: ===================================================================
starting test - test_stack_pt_randomization
Test Stack pointer randomization
stack pointer changed 56 times out of 64 tests
PASS - test_stack_pt_randomization
===================================================================
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

MaureenHelm, 2018-05-24:
Fixed by #7752 and #7838. Thank you @andrewboie!

closed 2018-05-24 by MaureenHelm