zephyrproject-rtos/zephyr #7704: nrf52_pca10040:tests/bluetooth/init/test_controller_dbg fails build with CONFIG_USERSPACE=y

andrewboie, 2018-05-21:
Very strange error on in nrf52_pca10040:tests/bluetooth/init/test_controller_dbg:

```
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/subsys/bluetooth/host/smp.c: In function ‘sign_test’:
/home/buildslave/src/github.com/zephyrproject-rtos/zephyr/subsys/bluetooth/host/smp.c:3864:1: error: r7 cannot be used in asm here
 }
 ^
zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/build.make:254: recipe for target 'zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/smp.c.obj' failed
make[3]: *** [zephyr/subsys/bluetooth/host/CMakeFiles/subsys__bluetooth__host.dir/smp.c.obj] Error 1
```

andrewboie, 2018-05-24:
I believe I know what this is.
Most ARM use the latest AAPCS which does not keep a frame pointer. There is impossibly convoluted logic and the exidx section used for stack unwinding.
I suspect this board uses the older version, where r7 is used to store the frame pointer, and GCC does not like this. Might have to do this directly in assembly.

carlescufi, 2018-05-28:
@andrewboie where is the aapcs version selected in the board? I don't think this does anything differently than other boards, the difference is perhaps that we are the only ones to build the Bluetooth host and controller together. 

CC @pizi-nordic

pizi-nordic, 2018-05-29:
The AAPCS designated r11 as a frame pointer, so we are still in the fog. I will try to reproduce this issue and try to find the cause.

pizi-nordic, 2018-05-29:
I traced it down.

1. The AAPCS defines `r11` as a frame pointer for long mode, however in thumb a `r7` is used (as lower registers might be used with more instructions).

2. The 09a8810b339b8266f835ea29f863a9cb825e486a adds the `r7` to the clobber list of  `_arch_syscall_invoke*()` calls. The `r7` must be listed there as it is overwritten by the `_do_syscall`.

3. The GCC prints the mentioned error if `r7` is used by assembly code in the function that has frame pointer.

So, as long as there is no frame pointer (-O1 or higher or -fomit-frame-pointer passed and no local data allocated on stack) in the function, everything will work OK. However a more complex code (using frame pointer, like the mentioned `sign_test`) will be not able to invoke any syscall.

IMHO this the code introduced in 09a8810b339b8266f835ea29f863a9cb825e486a should be updated in a way which ensure r7 preservation during syscall.

agross-linaro, 2018-05-29:
Yeah I'll take shortly.  We moved from using stack to using regs for these due to unaligned stack issues.  I didn't know the thumb mode did things a little different.

agross-linaro, 2018-05-30:
So the problem with fixing this I think goes back to the way args are passed.  So originally we used some stack space to store parameters to avoid using the r4/r5/r6/r7.  This didn't work out well due to the compiler sometimes creating unaligned stacks which threw off our stack retrieval of parameter information.

So to get around that issue, we move everything to use r0/r1/r2/r3/r4/r5/r6/r7 to pass information.  Clearly that isn't going to work in some cases.

So what if we instead pass in stack ptr to the syscall that points directly to the syscall frame that describes the parameters.  This is like the original solution but works around the stack padding that may or may not occur due to the alignment of the stack.  So we'd only use r0/r1/r2/r3 at most to pass parameters.  Any additional ones would rely on stack storage and the passing of the stack ptr in r0, r1, r2, or r3.

pizi-nordic, 2018-05-30:
The problem is that this calling convention uses r7. So for example the r0/r1/r2/r3/r4/r5/r6/r12 should work and you will not need to create stack area for arguments. What do you think?

agross-linaro, 2018-05-30:
R7 is for the saved link register.  We need that to be able to return back to the original caller.

pizi-nordic, 2018-06-01:
Summarising:
- We have to remove r7 from clobber list.
- To do so, we have to preserve r7 during sycall. At the moment the r7 holds saved LR (set in `_do_syscall`, used in `_arm_do_syscall`). 

I have created a PR https://github.com/zephyrproject-rtos/zephyr/pull/8103, which fixes this problem by using r8 instead of r7 to hold saved LR.

PR #8103 (2018-06-01): arm: userspace: Do not overwrite r7 during syscall.
The r7 register is used as a frame pointer on ARM Thumb. As result, it cannot be modified
by the assembly code in functions using stack frame.

This pull requests replaces r7 by r8, which is a general purpose register, fixes #7704.

All tests with CONFIG_USERSPACE=y passed without any problems.
commit 997a49ad (2018-06-01): arm: userspace: Do not overwrite r7 during syscall.

The r7 register is used as a frame pointer on ARM Thumb. As result, it
cannot be modified by the assembly code in functions using stack frame.

This commit replaces r7 by r8, which is a general purpose register.
Also it fixes #7704.
closed 2018-06-01 by nashif