zephyrproject-rtos/zephyr #7706: ARM hardware stack overflows do not report _NANO_ERR_STACK_CHK_FAIL or provide MPU fault information

andrewboie, 2018-05-22:
When stack guard pages are enabled and a thread overflows its stack, the error reported is as follows, causing tests/kernel/fatal to work improperly:

```
test stack overflow - timer irq
***** BUS FAULT *****
  Executing thread ID (thread): 0x2000002c
  Faulting instruction address:  0x1b6e
  Imprecise data bus error
Caught system error -- reason 0

    Assertion failed at /home/apboie/projects/zephyr/tests/kernel/fatal/src/main.c:224: test_fatal: )
bad reason code got 0 expected 2
```

Reason code 0 is _NANO_ERR_CPU_EXCEPTION, the expected code of 2 is _NANO_ERR_STACK_CHK_FAIL.

We need to improve this:
* There is logic in fault.c for dumping out MPU faults, but it is never called even though this is an MPU access violation. The user will not know that the fault was due to illegal memory access.
* If the MPU fault was due to access violation within the region used for the stack guard, the reason code provided to _NanoFatalErrorHandler should be _NANO_ERR_STACK_CHK_FAIL and not _NANO_ERR_HW_EXCEPTION.

I'm committing a FIXME workaround to the fatal test until this is fixed.

andrewboie, 2018-05-22:
per @MaureenHelm this may be specific to the NXP MPU

MaureenHelm, 2018-05-22:
The NXP MPU throws a bus fault on an access violation, whereas the ARM MPU throws a MemManage fault. Running this test on the mimxrt1050_evk board, which has the ARM MPU, gives the result:
```
test stack overflow - timer irq
posting 2048 bytes of junk to stack...
***** MPU FAULT *****
  Executing thread ID (thread): 0x2000002c
  Faulting instruction address:  0xb7c
  Data Access Violation
  Address: 0x20001000
Caught system error -- reason 0
```

ioannisg, 2018-05-23:
If I understand correctly, @andrewboie , _NANO_ERR_HW_EXCEPTION is the sole error code used in arm's fault.c, so this does not look like only an NXP issue, right?

MaureenHelm, 2018-05-23:
> If I understand correctly, @andrewboie , _NANO_ERR_HW_EXCEPTION is the sole error code used in arm's fault.c, so this part of the problem does not look like it is only an NXP issue, right?

Right. The NXP-specific issue is @andrewboie 's first bullet about not dumping MPU faults on parts like k64f that have the NXP MPU, because they take a bus fault instead. Parts like i.mx rt1050 that have the ARM MPU do dump MPU faults as expected.

In both cases, the reason code is still 0.

ioannisg, 2019-02-05:
Should we try to get this into 1.14, @MaureenHelm ? I might have some time for that, but only after Friday (bu that is ok for bug fixes, right?)

andrewboie, 2019-02-05:
we have plenty of time to address bug fixes for the LTS

andrewboie, 2019-02-08:
we tripped over this again in #13166

PR #13334 (2019-02-28): Arch arm core fix fault report nxp mpu
Fixes #7706 
Fixes #13342 
Closes #13432 

This PR
- investigates stack corruptions, not only or precise memory violations, but also on stacking errors
-  refactors fault.c, so the assessment of stack fail is moved to thread.c, as it accesses kernel structures,
- allows reporting thread stack corruption for user threads stacks,
- improves error messages and inline documentation related to ARM memory faults,
closed 2019-02-28 by andrewboie