zephyrproject-rtos/zephyr #7709: native_posix: hello_world fails to link on Fedora 28

jhedberg, 2018-05-22:
I've got a 32-bit libc and its headers installed. Compilation works, but hello_world fails at the linking stage:

```
[100%] Linking C executable zephyr.elf
/usr/bin/ld: Warning: size of symbol `zephyr_app_main' changed from 86 in ../libapp.a(main.c.obj) to 30 in kernel/libkernel.a(init.c.obj)
/usr/bin/ld: i386 architecture of input file `CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `../libapp.a(main.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(thread_entry.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(crc16_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(crc8_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(mempool.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(rb.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(printk.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(configs.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(native_posix_console.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(sys_clock_init.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libzephyr.a(native_posix_timer.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/soc/inf_clock/libarch__posix__soc__inf_clock.a(soc.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/core/libarch__posix__core.a(cpuhalt.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/core/libarch__posix__core.a(fatal.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/core/libarch__posix__core.a(posix_core.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/core/libarch__posix__core.a(swap.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `arch/posix/core/libarch__posix__core.a(thread.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `boards/boards/posix/native_posix/libboards__posix__native_posix.a(hw_models_top.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `boards/boards/posix/native_posix/libboards__posix__native_posix.a(timer_model.c.obj)' is incompatible with i386:x86-64 output
...
```

jhedberg, 2018-05-22:
@andrewboie I hear you've been using native_posix on Fedora 28 successfully? Do you have any idea what might be causing my linker failure?

andrewboie, 2018-05-22:
@jhedberg I have also been filing bugs on FC28 and native_posix see #7610 
I haven't seen this issue before.

aescolar-ot, 2018-05-23:
@jhedberg Can you confirm if after installing all the packages from the doc. you still have the issue?

jhedberg, 2018-05-23:
Turns out I was missing libstdc++-devel.i686. After installing it I'm able to build successfully. Sorry for the noise.

closed 2018-05-23 by jhedberg