zephyrproject-rtos/zephyr #7722: [Coverity CID :186049] Memory - corruptions in /samples/subsys/mgmt/mcumgr/smp_svr/src/main.c

mandarcthorat1, 2018-05-22:
Static code scan issues seen in File: /samples/subsys/mgmt/mcumgr/smp_svr/src/main.c
 Category: Memory - corruptions
 Function: main
 Component: Samples
 CID: 186049
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

carlescufi, 2018-05-23:
@nvlsianpu can you please take a look at this one?

nvlsianpu, 2018-05-23:
sure

nvlsianpu, 2018-05-23:
issue is false positive - memory corruption doesn't happen - just function for clearing the statistic data (`void
stats_reset(struct stats_hdr *hdr)`) doesn't know the peculiar structure of  statistic group (and can not know as the function is generic).

lpereira, 2018-05-23:
Closing as false positive.  Marked the issue in Coverity as such as well.

closed 2018-05-23 by lpereira