zephyrproject-rtos/zephyr #7733: [Coverity CID :186037] Memory - corruptions in /lib/posix/fs.c

mandarcthorat1, 2018-05-22:
Static code scan issues seen in File: /lib/posix/fs.c
 Category: Memory - corruptions
 Function: readdir
 Component: Other
 CID: 186037
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

ramakrishnapallala, 2018-05-23:
This can be closed as false positive.
code:
rc = strlen(fdirent.name);
memcpy(pdirent.d_name, fdirent.name, rc <= PATH_MAX ? rc : PATH_MAX);

strlen always return <= ARRAY_SIZE(fdirent.name). Here we making sure rc is not more than PATH_MAX.
So ideally we should never ever have case where rc > PATH_MAX unless the ARRAY_SIZE(fdirent.name) or strlen return value which is more than PATH_MAX.

lpereira, 2018-05-23:
Just have to make sure that the string is NUL-terminated, because that memcpy() can clip the string.

ramakrishnapallala, 2018-05-29:
OK

PR #7985 (2018-06-01): lib: posix: Make sure the name string is NULL terminated
Make sure the name string is NULL terminated in the readdir().

CID: 186037

Fixes #7733
closed 2018-06-01 by nashif