zephyrproject-rtos/zephyr #7755: getchar sample not working on nRF5x

carlescufi, 2018-05-22:
The commit https://github.com/zephyrproject-rtos/zephyr/commit/05c45e359a94e96353dfac1f0abeccf6893e988a#diff-fb209e20d9cf9e70b3dbd2748ae43f90 broke the `samples/subsys/console/getchar` sample.

carlescufi, 2018-05-22:
@jarz-nordic can you make sure your PR fixes this?

pfalcon, 2018-05-23:
> This commit 05c45e359a94e96353dfac1f0abeccf6893e988a https://github.com/zephyrproject-rtos/zephyr/pull/7431 broke the samples/subsys/console/getchar sample.

There's a comment on that commit that zephyr shell is broken too:
https://github.com/zephyrproject-rtos/zephyr/commit/05c45e359a94e96353dfac1f0abeccf6893e988a#commitcomment-29012316 . 

I don't have a nRF5x system with UART connection, so can't check it myself.

carlescufi, 2018-05-23:
@pfalcon different issue though. Both shell and getchar work with this PR anyway, so this is fine

PR #7960 (2018-05-28): drivers: serial: nrf: Fix interrupts enabling/disabling
This patch corrects the way the INTENSET and INTENCLR registers are
accessed. When these registers are written with a bitmask, specified
bits are set or cleared in the INTEN (interrupt enabling) register,
so there is no need to read the previous state of these registers
(when they are read, the current state of INTEN is returned).
For INTENSET this patch eliminates only one unnecessary read, but for
INTENCLR the change is crucial because in the previous version any
write to the register disabled all the enabled interrupts.

Fixes #7755.
PR #7681 (2018-06-12): drivers: uart: Refactor nrf uart shim
Replace all register defines and calls with Nordic nrfx HAL.

Fixes #7755
closed 2018-05-28 by carlescufi