zephyrproject-rtos/zephyr #7764: Implement posix `time()`

d3zd3z, 2018-05-22:
Several cryptographic operations require knowledge of the current time. Zephyr has a few mechanisms for tracking time, including `k_uptime_get()` and the beginnings of an RTC framework. There is also support for SNTP, which allows a network query of the current time.

The mbed TLS framework is set up to use the POSIX `time` call, by change a few macros in its config file. This issue seeks to implement this call.

Because newlib already contains a `time` function (that fails), I suggest calling this `k_time()` to fit with other Zephyr APIs. A configuration option would determine if this is based off of a RTC, or if the value comes from `k_uptime_get()` with an offset. It will also need a call to set the current time (likely based on the result of SNTP.

pfalcon, 2018-05-23:
Why not just implement the POSIX time() call (i.e. make sure that newlib's implementation works as expected)?

d3zd3z, 2018-05-24:
Looks like there is already a clock_gettime() in the posix code that works for the monotonic time, so implementing this for CLOCK_REALTIME is a start. I'm not sure what the newlib implementation is doing, and maybe it can be fixed.

d3zd3z, 2018-07-09:
I have verified that if `clock_gettime()` is implemented correctly, the newlib `time()` does work correctly.