zephyrproject-rtos/zephyr #7766: tests/subsys/fs/fat_fs_api - fat_fs_basic_test hangs in fatfs_mount

pswarnak, 2018-05-22:
The test **tests/subsys/fs/fat_fs_api** hangs without any console log or error message.

Arch: x86
Board: arduino_101

Console Log:
```
Booting Zephyr OS v1.12.0-rc1
Running test suite fat_fs_basic_test
===================================================================
starting test - test_fat_mount

```

Tried further debugging, the test hangs in fatfs_mount function in the file "subsys/fs/fat_fs.c"
This is the exact line of code:
`res = f_mount((FATFS *)mountp->fs_data, &mountp->mnt_point[1], 1);
`

The test was working fine till zephyr-v1.11.0 tag. The sub testcase(test_fat_mount) was recently added as part of VFS change.

MaureenHelm, 2018-05-25:
@ramakrishnapallala Any update?

ramakrishnapallala, 2018-05-25:
I think the issue is with SPI flash driver. I will debug and update the status accordingly.

ramakrishnapallala, 2018-05-28:
As expected, the issue is with drivers/flash/spi_flash_w25qxxdv.c
system got stuck in  a while loop in wait_for_flash_idle() function. This, in turn, making a call spi_transceive() and expects a register bit to be cleared(W25QXXDV_WIP_BIT) which is not happening.

ramakrishnapallala, 2018-05-28:
looks like the following commit introduced the issue, even the test under samples/drivers/spi_flash/ is also failing at the same place. I need to check with Tomasz on this one.

commit 16cb7ab883332fce0b6075c0e840c1919db23862
Author: Tomasz Bursztyka <tomasz.bursztyka@linux.intel.com>
Date:   Tue Jan 30 15:01:36 2018 +0100

    drivers/flash: Switch W25QXXDV driver to new SPI API

    This saves quite some memory.

    Signed-off-by: Tomasz Bursztyka <tomasz.bursztyka@linux.intel.com>

ramakrishnapallala, 2018-05-28:
CC @tbursztyka 

Another observation is, the original tests are using QMSI SPI controller driver but the latest code uses DW SPI driver and removes the QMSI drivers.

@tbursztyka can we make samples/drivers/spi_flash/ work with the latest code?

ramakrishnapallala, 2018-05-29:
@tbursztyka on the while loop part, I am with you. I can fix that part.

ramakrishnapallala, 2018-05-29:
@tbursztyka good to hear that you found the issue. Thanks.

tbursztyka, 2018-05-29:
actually there were 2 issues. The other issue is the use of tx buf/len NULL/>0 is found nowhere but in w25qxxdv driver, and that case was missing in dw driver. 

@ramakrishnapallala please test #8003 I verified it on my arduino 101 and it works just fine.

closed 2018-05-29 by nashif

pswarnak, 2018-05-31:
**Status Update:**
#8003 resolves test_fat_mount hang issue.
However, we the main test case "fat_fs_basic_test" still results into Assertion failure in sub test cases test_fat_file and test_fat_dir.
Issue reported in https://github.com/zephyrproject-rtos/zephyr/issues/8065.

PR #8003 (2018-05-29): Spi flash fixes
fixes #7766
commit af411f31 (2018-05-29): boards/x86: Fix support for w25qxxdv on arduino_101

It requires to use w25qxxdv spi gpio cs settings.

Fixes #7766