zephyrproject-rtos/zephyr #7780: Using latest Openthread in Zephyr

akansal1, 2018-05-23:
We are considering openthread with zephyr in our implementation but openthread commit in Zephyr is dated Feb 14, 2018 post which multiple changes have been made on the project.

I want to know if it is possible to build Zephyr with latest commit openthread ? If yes, then what may be challenges that we might face?

thanks

SebastianBoe, 2018-05-23:
FYI: @KaSroka

rlubos, 2018-05-23:
Hi,
OpenThread version used by Zephyr is specified here: https://github.com/zephyrproject-rtos/zephyr/blob/fcc56e315f1dcc0c25422cb528662fe8dfe0b059/subsys/net/lib/openthread/CMakeLists.txt#L26

You can try to set newer commit here and then regenerate project and rebuild. Be aware that you might encounter compilation errors in case OT or platform API changed. An example of commit that bumps OT version here: https://github.com/zephyrproject-rtos/zephyr/commit/c0af4de7b3c3cec4c2f632493ad12e9e50a55b73

I did a quick test with the most recent OT commit for today and luckily for you OT samples in Zephyr compile nicely without any warnings/errors. I did not check if they actually work though so it's up to you to verify it.

akansal1, 2018-05-23:
Thank you @rlubos. 

Will this be the standard practice that we have to do, to keep intergrated external libraries up to date on zeyphr?

rlubos, 2018-05-24:
I'm not aware on any plans to change the mechanics of how OpenThread is integrated into Zephyr, so at least for some time this will be a valid way to update OT version.

closed 2018-05-24 by MaureenHelm