zephyrproject-rtos/zephyr #7793: samples\sensor: bme280 and tmp112 hangs without any console output on quark_se_c1000_devboard

pswarnak, 2018-05-23:
The test **samples\sensor\bme280** hangs without any console log or error message.

**Arch:** x86
**Board:** quark_se_c1000_devboard
**Zephyr tag:**  v1.12.0-rc1 - commit e7509c17c5938b48f9ad4b7ae78199bb0faa6113.
**Pre-requisite:** Sensor connected on I2C SCL, SDA pins of x86 core.

**Error:**
*No console log available at all.*

The test was working fine till zephyr-v1.11.0 tag.

**Steps to reproduce:**
```
rm -rf build && mkdir build && cd build
cmake -D BOARD=quark_se_c1000_devboard ../
make BOARD=quark_se_c1000_devboard flash

Check the console log.
```

**Expected output:**
```
***** Booting Zephyr OS v1.12.0-rc1 *****
dev 0xa8001354 name BME280
temp: 23.440000; press: 64.988484; humidity: 70.857421
temp: 23.440000; press: 64.988593; humidity: 70.857421
```

**Additional Info:**
Same issue seen for samples\sensor\tmp112 sensor testcase too.
quark_se_c1000_devboard has TMP112 sensor internally, does not require external sensor to be connected on x86 core.

nashif, 2018-05-24:
> The test was working fine till zephyr-v1.11.0 tag.

and why is this being reported now?

pswarnak, 2018-05-28:
> and why is this being reported now?

That was the last tag that i had compared with. I could not get sufficient time to check the exact git commit, where these tests started failing. I will try to provide the exact commit for ease.
I had reported [7026](https://github.com/zephyrproject-rtos/zephyr/issues/7026) for sensor failure that was fixed by @SavinayDharmappa 
This is second sensor related issue reported after zephyr-v1.11.0 tag.

pswarnak, 2018-05-28:
Figured out the exact git commit resulting into test failure using "git bisect".

The test cases are failing after the commit 74d1835 by @tbursztyka.
@nashif: This commit was merged as part of [PR-7623](https://github.com/zephyrproject-rtos/zephyr/pull/7623)  on May 18, 2018 and reported by us on May23, 2018. Hope, this answers your query.

Sensor test case passed by reverting the commit.
Please let us know, if test case require modification to make it run with your committed changes.

closed 2018-05-28 by nashif

pswarnak, 2018-05-31:
Thanks @punitvara. Validated the test, works fine on 1.12.0-rc2 tag.

PR #7955 (2018-05-28): dts: x86: Fix wrong interrupt number for I2C
Wrong interrupt number was populated from dts. This
commit solves following issue:

Fixes #7793