zephyrproject-rtos/zephyr #7804: samples/sensor/apds9960 fails with assertion "Fatal fault in essential thread" on quark_se_c1000_ss_devboard

pswarnak, 2018-05-23:
The test **samples/sensor/apds9960** results into exception on arc core.

**Arch:** arc
**Board:** quark_se_c1000_ss_devboard
**Zephyr tag:**  v1.12.0-rc1 - commit e7509c17c5938b48f9ad4b7ae78199bb0faa6113.
**Pre-requisite:** Sensor connected on I2C SCL, SDA pins of arc core.

**Error:**
```
****** Booting Zephyr OS v1.12.0-rc1-41-gfcc56e3 *****
APDS9960 sample application
Exception vector: 0x3, cause code: 0x4, parameter 0x0
Address 0x40002248
Current thread ID = 0xa8000508
Faulting instruction address = 0x40002248
Fatal fault in essential thread! Spinning...
```

The test was working fine till zephyr-v1.11.0 tag.

**Steps to reproduce:**
```
rm -rf build && mkdir build && cd build
cmake -D BOARD=quark_se_c1000_ss_devboard ../
make BOARD=quark_se_c1000_ss_devboard flash

Check the console log.
```

**Expected output:**
```
***** BOOTING ZEPHYR OS v1.11.0 - BUILD: May 23 2018 20:29:24 *****                                                   
APDS9960 sample application                                                                                           
ambient light intensity without trigger is 0
proxy without trigger is 42
ambient light intensity without trigger is 211
```

MaureenHelm, 2018-05-25:
@punitvara Any update?

punitvara, 2018-05-28:
@MaureenHelm  Fixed

pswarnak, 2018-05-31:
Thanks @punitvara. Validated the test, works fine now.