zephyrproject-rtos/zephyr #7812: tests: Crypto tests fail on nrf52 boards after enabling user mode

Mierunski, 2018-05-23:
Some tests started failing after commit: efe0c4b7647c5e5aa53af47679bb8625fb387b19 on boards nrf52_pca20020 and nrf52_vbluno52 , as can be seen [here](https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/15577/3/tests)

Error message is 
```
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: zephyr/app_sizing_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: region `SRAM' overflowed by 137508 bytes
```

It can be checked by building 'tests/crypto/sha256' for nrf52_pca20020 board, source of issue was determined by bisection.

galak, 2018-05-23:
This in theory should be fixed with #7827

andrewboie, 2018-05-23:
please re-test should be good now.

closed 2018-05-24 by andrewboie