zephyrproject-rtos/zephyr #7818: big_http_download stuck during download on qemu_x86

rlubos, 2018-05-23:
After receiving first packet with the downloaded package application get's stuck and keeps idling. I'm using current Zephyr master with default configuration on qemu_x86. I attach screenshot with logs from Wireshark, there is some mess due to TCP window size, but I guess it is correct in case of small window? Not sure if am I doing something wrong or if there is an problem with the application. Any ideas what could be wrong?

[image]

pfalcon, 2018-05-23:
> or if there is an problem with the application.

I'd like to start with a methodological point. The question above, taken literally, is very easy to answer: all socket samples shipped with Zephyr to date can be built on a POSIX system (besides Zephyr). Just do:

~~~
make -f Makefile.posix
~~~

on a Linux system and you'll see that there's nothing wrong "with the application", it just works. Thus, we can strike that thru and proceed with posing next questions, specifically:

2. Is there a problem with a small, thin shim which implements BSD Sockets API on top of native Zephyr API?
3. Is there a problem with Zephyr IP stack?

This very simple, just as very important, principle was downplayed and rejected when discussing https://github.com/zephyrproject-rtos/zephyr/pull/7118. Users of that "API" won't have a freedom to triage application vs stack issues so easily. (Without introducing head-over-heels tricks, which I'm sure will be introduced.)

Anyway, not our case here, and we can concentrate on qq. 2 and 3 here.

pfalcon, 2018-05-23:
For reference, my Wireshark window:

[image]

rlubos, 2018-05-24:
Thank you for the feedback. In no means I wanted to blame the "sample" part of the application, by application I meant entire solution, perhaps I should revisit my vocabulary.

Anyway, I did some investigation and it appears that the root cause for the issue described is handling of Zero Window Probe, that can be seen in the Wireshark log provided. Apparently TCP implementation on the server I was connecting to, as well as Wireshark's one expect, that after zero window length was notified by the peer, it will drop any data received. Therefore the server ignored the ACK sent by Zephyr app, which indicated that byte was consumed and stopped transfering data.

Whether this is correct behaviour or not is something to discuss, I saw there are different opinions in the web (for example [here](https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=10745)). Anyway, as a proof of concept I have a fix that was verified to make communication work again (https://github.com/rlubos/zephyr/tree/zero-window-probe-handling). It's up to discussion whether we want to handle this corner case in Zephyr.

Other solution helping with this issue was doubling TCP window size, so we're simply not hitting this Zero Window Probe problem. It's more like a workaround, but it helped as well. It would require some changes as well, as currently TCP window size is not something configurable (which I think should be though, as it's tighly related to nuber of buffers we have, and that is configurable).

pfalcon, 2018-05-24:
> the root cause for the issue described is handling of Zero Window Probe, that can be seen in the Wireshark log provided

Right, so the next step in my queue is submitting a ticket titled "net: Zero Windows Probes are not handled properly", and discussing the current behavior and how it should work actually. Feel free to beat me on submitting such a ticket ;-).

pfalcon, 2018-05-24:
The handshake uses packets of the same size, except I receive from the peer SYN/ACK with MSS option = 1440, whereas @rlubos - 1460. We don't send MSS option. So, how it happens that based on this, in @rlubos case, the peer selects MSS of 1280, while in my case - of 640, is a mystery.

pfalcon, 2018-05-24:
Ah, right MSS is asymmetrical actually...

pfalcon, 2018-05-24:
Even by sending explicit MSS option in SYN, I can't get peer to send anything else but 640 bytes data segments...

pfalcon, 2018-05-24:
@rlubos , can you tell what version of Wireshark you use (and on what distro). I have a feeling that mine simply can't decode Zero Window Probe, as it decodes as TCP Keep-Alive even obvious retransmits. Mine is 2.2.6 on ubuntu 16.04.

pfalcon, 2018-05-24:
Then I tried my linux localhost, then google.com. It's all the same - I don't ZeroWindowProbe like on @rlubos' screenshot, instead I see Keep-Alive's.

pfalcon, 2018-05-24:
Then I find this: https://www.ietf.org/mail-archive/web/tcpm/current/msg08397.html . So, that thread says that indeed Linux uses Keep-Alive's for the purpose of zero window probing, while other systems, like FreeBSD and (as referenced elsewhere on the net) MS Windows.

Consequently, @rlubos, I don't know how to explain your case. Is there a chance that you have a transparent proxy or NAT using those systems, which go as far as actually reformat TCP packets instead of just forwarding? Another explanation could be that archive.ubuntu.com has such a server behind a load balancer, and in many times I ran big_http_download I never hit it, while you did - once, all the time - you don't say whether this issue reported is 100% reproducible for you.

pfalcon, 2018-05-24:
Then we can prepare following test setup:

~~~
--- a/samples/net/sockets/big_http_download/src/big_http_download.c
+++ b/samples/net/sockets/big_http_download/src/big_http_download.c
@@ -29,7 +29,7 @@
 
 /* This URL is parsed in-place, so buffer must be non-const. */
 static char download_url[] =
-    "http://archive.ubuntu.com/ubuntu/dists/xenial/main/installer-amd64/current/images/hd-media/vmlinuz";
+    "http://microsoft.com/";
 /* Quick testing. */
 /*    "http://google.com/foo";*/
 
diff --git a/subsys/net/ip/tcp.c b/subsys/net/ip/tcp.c
index fec12c550..c39205875 100644
--- a/subsys/net/ip/tcp.c
+++ b/subsys/net/ip/tcp.c
@@ -274,7 +274,7 @@ struct net_tcp *net_tcp_alloc(struct net_context *context)
 	tcp_context[i].context = context;
 
 	tcp_context[i].send_seq = tcp_init_isn();
-	tcp_context[i].recv_wnd = min(NET_TCP_MAX_WIN, NET_TCP_BUF_MAX_LEN);
+	tcp_context[i].recv_wnd = 40; //min(NET_TCP_MAX_WIN, NET_TCP_BUF_MAX_LEN);
 	tcp_context[i].send_mss = NET_TCP_DEFAULT_MSS;
 
 	tcp_context[i].accept_cb = NULL;
~~~

And with it, I indeed can reproduce the issue, and specific error logging happens on the console.
[image]

pfalcon, 2018-05-24:
Actually no, I don't see any problem here, except that Wireshark's "AI" seems to be misinterpreting ZeroWindowProbe. It treats it as if it was some kind of a magic packet, but as explained in https://github.com/zephyrproject-rtos/zephyr/issues/7876#issuecomment-391745233 , zero windows probe is actually just a normal TCP segment which can be interpreted by a receiver in one of the standard ways. This Wireshark issue is described e.g. in https://osqa-ask.wireshark.org/questions/21230/tcp-acked-unseen-segment-wireshark-bug , https://bugs.wireshark.org/bugzilla/show_bug.cgi?id=10745

With that in mind, what we have is:

* The host sent us 40 bytes, we replied that our window is full
* Then host sent ZeroWindowProbe with a new byte, we ACKed this byte, and said we have 39 more in window
* The host sent us 39 bytes, we acked them all, but as those bytes weren't consumes yet, we ended up saying that we have window of just 1.
* We can only imagine what happened next. Likely, the host got snobby about the proposed window, and decided not to talk to us again. It waited a bit (3s) to see if we have something more to say, but oops, Zephyr IP stack doesn't communicate window updates proactively. So, the host RSTed on us.

Summary of this is: Zephyr TCP implementation is rather poor, bare and lacks many features. Like communicating receive window updates. Note that eager communication of such is no-no either, because other each recv(sock, buf, 1) would result in a whole packet (ACK) being sent to peer. How decent (embedded) IP stacks, like lwIP, so that is to cache window updates - send them explicitly only if enough amount of free window accumulate, **or** when a timeout expires.

Decent embedded stacks (like lwIP) are also cooperative, which allows them to behave deterministically, reason about this behavior, and tweak it for performance. On the other hand, Zephyr IP stack is preemptive with lurking concurrency issues, so I'm not sure how to add new features to it before resolving the existing issues (or otherwise risking making them worse). I can however submit a ticket for this issue.

pfalcon, 2018-05-24:
As a final test, I'm applying following test changes:

~~~
--- a/samples/net/sockets/big_http_download/src/big_http_download.c
+++ b/samples/net/sockets/big_http_download/src/big_http_download.c
@@ -29,7 +29,7 @@
 
 /* This URL is parsed in-place, so buffer must be non-const. */
 static char download_url[] =
-    "http://archive.ubuntu.com/ubuntu/dists/xenial/main/installer-amd64/current/images/hd-media/vmlinuz";
+    "http://microsoft.com/";
 /* Quick testing. */
 /*    "http://google.com/foo";*/
 
diff --git a/subsys/net/ip/tcp.c b/subsys/net/ip/tcp.c
index fec12c550..c39205875 100644
--- a/subsys/net/ip/tcp.c
+++ b/subsys/net/ip/tcp.c
@@ -274,7 +274,7 @@ struct net_tcp *net_tcp_alloc(struct net_context *context)
 	tcp_context[i].context = context;
 
 	tcp_context[i].send_seq = tcp_init_isn();
-	tcp_context[i].recv_wnd = min(NET_TCP_MAX_WIN, NET_TCP_BUF_MAX_LEN);
+	tcp_context[i].recv_wnd = 40; //min(NET_TCP_MAX_WIN, NET_TCP_BUF_MAX_LEN);
 	tcp_context[i].send_mss = NET_TCP_DEFAULT_MSS;
 
 	tcp_context[i].accept_cb = NULL;
diff --git a/subsys/net/lib/sockets/sockets.c b/subsys/net/lib/sockets/sockets.c
index 5bcd771f5..5f672245e 100644
--- a/subsys/net/lib/sockets/sockets.c
+++ b/subsys/net/lib/sockets/sockets.c
@@ -443,7 +443,7 @@ static inline ssize_t zsock_recv_stream(struct net_context *ctx,
 	} while (recv_len == 0);
 
 	if (!(flags & ZSOCK_MSG_PEEK)) {
-		net_context_update_recv_wnd(ctx, recv_len);
+//		net_context_update_recv_wnd(ctx, recv_len);
 	}
 
 	return recv_len;
~~~

With it, I get the following console output:

~~~
***** Booting Zephyr OS v1.11.0-1619-g5b20350a7 *****
[net/app] [INF] net_app_init: Initializing network
[net/app] [INF] setup_dhcpv4: Running dhcpv4 client...
Preparing HTTP GET request for http://microsoft.com:80/
addrinfo @0x401040: ai_family=2, ai_socktype=1, ai_protocol=6, sa_family=2, sin_port=5000
sock = 4199520
[net/tcp] [ERR] _tcp_established: Context 0x00401460: overflow of recv window (0 vs 1), pkt dropped
[net/tcp] [ERR] _tcp_established: Context 0x00401460: overflow of recv window (0 vs 1), pkt dropped
~~~

in full accordance with the analysis described in https://github.com/zephyrproject-rtos/zephyr/issues/7876.

So, that's the only stable and reproducible issue on *our* side that I see, and which needs to be fixed. (The next issue on our side is "TCP handling is poor", but unfortunately one can't submit a few lines patch and have that fixed).

rlubos, 2018-05-25:
Thank you for detailed analysis.

> can you tell what version of Wireshark you use (and on what distro)

So I'm running Arch Linux (on a VMWare), Wireshark 2.6.0 

>  So, how it happens that based on this, in @rlubos case, the peer selects MSS of 1280, while in my case - of 640, is a mystery.

And vice versa, I've used the IP adress from you Wireshark log as a test yesterday, and still have the same behavior - 1280 MSS and failure on Zero Window probing. I guess we are connecting to different, mirrored servers as we are at different location?

>  Is there a chance that you have a transparent proxy or NAT using those systems

Well, I'm running on a VM and I'm in corporate network, I'm not sure what tricks do they do there.

> you don't say whether this issue reported is 100% reproducible for you

Right, my mistake - it happens all the time. I'm always connecting to the same server and it always reject the ACK for the zero window probe. Just tried with `microsoft.com` with the same result - ACK rejected. Works on #7867.

> Actually no, I don't see any problem here, except that Wireshark's "AI" seems to be misinterpreting ZeroWindowProbe.

Well for me it looks fine, the difference is that the server does not stall after receiving `ACK = SEQ + 1` in the repsonse, that confirms the packet was consumed. This just shows how different servers may behave differently.

pfalcon, 2018-05-25:
@rlubos , based on comments in https://github.com/zephyrproject-rtos/zephyr/pull/7867#issuecomment-392005562, can you try to reproduce this issue in other networking environments, e.g. at home, via 3g, etc.?

big_http_download exists since Feb 13, and before that, I was running it for a year as a MicroPython application: https://github.com/pfalcon/micropython/blob/zephyr-wip/ports/zephyr/scripts/test_dl.py . I never see stably reproducible issues like you report. And over this time, I run it from different ISP in different countries. Well, I just switched countries, and now the app works for me the same as it did a week ago.

I think the first step is figuring out what kind of IP stack sends you those ZWPs. As mentioned with a reference above, Linux isn't supposed to send len=1 ZWPs, it sends len=0 Keep-Alives instead.

pfalcon, 2018-05-25:
> So I'm running Arch Linux (on a VMWare), Wireshark 2.6.0
> Well, I'm running on a VM

To clarify, do you have a Windows machine, on which you run VMWare with Arch Linux inside, and in this VM, you run Zephyr development, including qemu_x86 with networking?

rlubos, 2018-05-25:
> To clarify, do you have a Windows machine, on which you run VMWare with Arch Linux inside, and in this VM, you run Zephyr development, including qemu_x86 with networking?

Yes. Windows 10 as host, Arch linux as a guest (running inside VMWare) and qemu_x86 in Arch. Any issues with that?

> can you try to reproduce this issue in other networking environments, e.g. at home, via 3g, etc.

Just tried with LTE through my phone. I've got similar results, though packets were smaller. Once I received ZWP, communication stalls. Though this time like 1 for 5 runs succedded - client never advertised zero window in that case, hence no probes.

I can try with different ISP and native Linux machine once I'm home.

pfalcon, 2018-05-25:
> Yes. Windows 10 as host, Arch linux as a guest (running inside VMWare) and qemu_x86 in Arch. Any issues with that?

So, that would be much plausible explanation, rather than transparent proxies/NATs, why you are talking to Linux host, but TCP behavior looks like Windows' - VMWare just sends guest's data using the host Windows stack.

> I can try with different ISP and native Linux machine once I'm home.

So, based on the above, a better test might to try to use natively running Linux system, if you have such possibility.

I from my side will think how to get good Windows testing system. If you know a server which runs Windows and talks plain HTTP, let me know ;-). (microsoft.com for example first thing redirects to https://, so HTTP replies are too small for good testing.)

rlubos, 2018-05-28:
> So, that would be much plausible explanation, rather than transparent proxies/NATs, why you are talking to Linux host, but TCP behavior looks like Windows' - VMWare just sends guest's data using the host Windows stack.

I think you'v hit the point here. As promised I did some experiments on my local machine, where I have both native Linux and Linux running in a VMWare. So it looks like indeed, that if network interface is configured as NAT in the VM, Windows is proxying the connection, therefore we talk to the Windows TCP/IP stack, not the actual server.

For the future reference, If anyone face this problem, switching to bridged network interface helps - allows VM to talk directly to the server.

I think that there still is an issue with ZWP when talking to Windows servers, as I observed the same probing mechanism when talking to different servers running Windows. I wasn't able to find a server with such behaviour that would actully send some reasonable amount of data - all that I tried just redirected to https. Perhaps I will give it a try and play a little bit with TLS connection.

So I think that for now we could close #7867 (and perhaps revive it when reproducible issue can be identified with real server), and focus on #7876 as suggested by @pfalcon.

nashif, 2019-02-19:
@pfalcon @rlubos is this still an issue?

rlubos, 2019-02-19:
After reconfiguring my VM I've not hit it ever again, I guess we can close it.

closed 2019-02-19 by rlubos