zephyrproject-rtos/zephyr #7819: minnowboard build breakage

galak, 2018-05-23:
The following change breaks building several configs:

`./scripts/sanitycheck -p minnowboard -s tests/benchmarks/footprint/benchmark.footprint.regular`

```
commit efe0c4b7647c5e5aa53af47679bb8625fb387b19 (refs/bisect/bad)
Author: Andrew Boie <andrew.p.boie@intel.com>
Date:   Mon May 21 12:44:38 2018 -0700

    tests: enable user mode by default
```

galak, 2018-05-23:
Seeing a number of failures related to this change:
https://app.shippable.com/github/zephyrproject-rtos/ci-test/runs/2036/summary/console

galak, 2018-05-23:
These 2 tests seem to be the majority of issue, in the size growing to big:
tests/crypto/sha256/crypto.sha256 
tests/crypto/hmac_prng/crypto.hmac_prng

andrewboie, 2018-05-23:
Looking at these now.

PR #7827 (2018-05-23): Test failures fixes
Enabling user mode exposed some problems with how ARC uses the STACK_ALIGN macro with user mode, some tests with very large stacks, and some footprint tests.

Fixes #7819
closed 2018-05-23 by galak