zephyrproject-rtos/zephyr #7848: CONFIG_BMM150_SET_ATTR not defined, giving dead code

ulfalizer, 2018-05-24:
`drivers/sensor/bmm150/Kconfig` has no definition for `BMM150_SET_ATTR`, which is set by `samples/sensor/bmm150/prj.conf`.

`drivers/sensor/bmm150/bmm150.c` has a bunch of `#if defined(BMM150_SET_ATTR)` (and some `#if defined(BMM150_SET_ATTR_REP)`), which are probably dead code. If those are meant to be Kconfig symbols, they'll need Kconfig definitions (and a 'CONFIG_' prefix in the `#ifdef`s).

ulfalizer, 2018-05-24:
CC @punitvara 

I'm trying to get rid of all assignments to undefined Kconfig symbols, so it can be turned into an error during testing.

ulfalizer, 2018-05-25:
I'm removing just the `CONFIG_BMM150_SET_ATTR` assignment in https://github.com/zephyrproject-rtos/zephyr/pull/7935. Remove the code as well if it isn't needed.

punitvara, 2018-05-28:
@ulfalizer  Extra code could be useful when interrupt functionality will be added. @ulfalizer @MaureenHelm  Do you suggest to remove extra code as of now ?

punitvara, 2018-06-06:
@ulfalizer  ??

ulfalizer, 2018-06-12:
@punitvara 
Sorry, missed it.

No panic to remove it on my part, if you think it might get used.

punitvara, 2018-06-12:
Ok no problem @ulfalizer Closing this issue as code is reusable to enable interrupt

closed 2018-06-12 by punitvara