zephyrproject-rtos/zephyr #7858: Test for k_thread_foreach() is failing only on DesignWare(R) ARC(R) EM Starter Kit

spoorthik, 2018-05-24:
**Test location:** tests/kernel/profiling/profiling_api

**Test Identifier:** kernel.profiling.k_thread_foreach

**Execution log:**
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.12.0-rc1-115-ge9174c6 (delayed boot 1000ms) *****
Running test suite profiling_api
===================================================================
starting test - test_call_stacks_analyze_main
 from main thread:
 Test (real size 2304):  unused 2116 usage 188 / 2304 (8 %)
 Test (real size 2304):  unused 2200 usage 104 / 2304 (4 %)
 Test (real size 2304):  unused 2280 usage 24 / 2304 (1 %)
 Test (real size 2304):  unused 1980 usage 324 / 2304 (14 %)
 PASS - test_call_stacks_analyze_main
 ===================================================================
 starting test - test_call_stacks_analyze_idle
 from idle thread:
 PASS - test_call_stacks_analyze_idle
 ===================================================================
 starting test - test_call_stacks_analyze_workq
 from workq:
 Test (real size 2304):  unused 2092 usage 212 / 2304 (9 %)
 Test (real size 2304):  unused 2056 usage 248 / 2304 (10 %)
 Test (real size 2304):  unused 2136 usage 168 / 2304 (7 %)
 Test (real size 2304):  unused 1980 usage 324 / 2304 (14 %)
 Test (real size 2304):  unused 2092 usage 212 / 2304 (9 %)
 Test (real size 2304):  unused 2056 usage 248 / 2304 (10 %)
 Test (real size 2304):  unused 2136 usage 168 / 2304 (7 %)
 Test (real size 2304):  unused 1980 usage 324 / 2304 (14 %)
 PASS - test_call_stacks_analyze_workq
 ===================================================================
 starting test - test_k_thread_foreach
 TEST: Newly added thread found
 TEST: tid: 0x800078ac, prio: 0

     **Assertion failed at zephyr.git/tests/kernel/profiling/pr
 thread_callback() not getting called
 FAIL - test_k_thread_foreach**
 ===================================================================
 ===================================================================
 PROJECT EXECUTION FAILED

**Recent commit in which the failure observed:** (the commit may not be the root cause of the issue)
e9174c6ddeddf58d8aebba5c89e50965731899e6

ruuddw, 2018-05-24:
Thanks for testing, what configuration was tested? For em_starterkit_em7d, with MPU and userspace support, I saw a similar fail. For non-MPU configuration like em9d I believe the test passes.
PR #777 is work in progress fixing the userspace issues.

andrewboie, 2018-05-25:
This is failing on ARM sam_e70_xplained as well

andrewboie, 2018-05-25:
@ramakrishnapallala can you please look at this?

spoorthik, 2018-05-25:
@ruuddw Yes, the test fails on em_starterkit_em7d with MPU enabled.
@andrewboie I didn't find failures on sam_e70_xplained though. May be it fails on any specific configuration?

vonhust, 2018-05-25:
@andrewboie @ruuddw  , it's not a real issue for ARC. In ARC user space support, for kernel thread, to fully utilize the stack,  the privilege stack is merged with user declared stack, i.e. the thread->stack_info.size ==  CONFIG_PRIVILEGE_STACK_SIZE + STACKSIZE.   if we change

`(thread->stack_info.size == STACKSIZE)` to  `(thread->stack_info.size >= STACKSIZE)` .

The test will pass.

Considering other arch may also do like like (I think it's meaningful to not waste privilege stack for kernel thread),  I suggest to update the test as described above.

andrewboie, 2018-05-25:
@spoorthik I just built it in-tree, ran 'make flash' and observed the same error on sam_e70_xplained.

ramakrishnapallala, 2018-05-25:
Looking into the issue.

ramakrishnapallala, 2018-05-27:
Ran the test :  tests/kernel/profiling/profiling_api on sam_e70_xplained and frdm_k64f boards. The test is passing. The only issue I see is,  after the test gets executed i see continuous stack dumps due to the _sys_soc_suspend call in the test (main.c). This is behavior should be in line with old API as well as it depends in suspend/resume call flows.

This can be fixed by conditionally running the test based on a test variable.

ramakrishnapallala, 2018-05-28:
able to reproduce the failure case on sam e70. reason for failure is the actual stack size allocated (thread->stack_info.size = 480) is less than what is requested (512) due to the following flags:

CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT
CONFIG_USERSPACE

I am modifying the test as per @vonhust suggestion.

andrewboie, 2018-05-30:
There's an assumption that thread->stack_info.base / .size represents the bounds of the actual stack buffer being used. For instance the stack analysis code needs it to be like this.
I need to spend some time today looking at the ARC/ARM code to see what is going on.

ramakrishnapallala, 2018-06-01:
fix merged.

closed 2018-06-01 by ramakrishnapallala